qmk/qmk_firmware #8010: Suggestion: Analog to Digital Analog Support

DeadlyDad, 2020-01-25:
I just heard about this project and, after searching the documentation, found that it doesn't have a feature that I am using in a project of mine.  I am using a Wii Nunchuck with an ATMEGA32u4 as a USB _keyboard_ controller for a particular application.  It will be using PWM to output the analog values from the joystick as cursor key input, thus allowing **any** application to have simulated analog controls.  (I used the same concept years ago to use my force feedback steering wheel with even _Flash_ racing games.  When used with a proper racing game, the 'ADAC' (analog -> 'digital analog' controller) system worked *exactly* the same as when in full wheel mode.)

My suggestion is that y'all add PWM output to your firmware.  Any microcontroller with ADC could support it.  Obviously, there would need to be provision for turning it off and on, whether through a switch, using dead zones, or something else.  (Imagine how handy it would be to have a nub in the middle of your keyboard that allowed for moving the mouse cursor around at different rates depending on how far from center it was moved, without needing to emulate a mouse as well.)

Hope this helps!
Keith Olson

closed 2020-05-25 by stale[bot]

inigosola, 2021-05-14:
> I used the same concept years ago to use my force feedback steering wheel with even _Flash_ racing games

What about this?

DeadlyDad, 2021-05-15:
I'm sorry; what are you asking?