qmk/qmk_firmware #8017: planck/ez build failed

muukii, 2020-01-26:
I'm trying to create my settings with qmk_firmware.
Before I used Oryx visual configurator.

And I faced following build error from make command.
Does anyone have ways to solve this problem?

```
$ make planck/ez:muukii
QMK Firmware 0.7.120
Making planck/ez with keymap muukii

lib/lufa/LUFA/Build/LUFA/lufa-sources.mk:32: *** Makefile ARCH option cannot be blank.  Stop.
/bin/sh: .//.build/error_occurred: No such file or directory
make: *** [planck/ez:muukii] Error 1
```

drashna, 2020-02-03:
run `make git-submodule`. 

Otherwise, I can't reproduce this.

muukii, 2020-02-03:
@drashna 
Thanks!
Actually, I tried this command. But it's still stuck.

but, I can forward the next step, I changed the other tag from mater branch.

closed 2020-02-08 by muukii