qmk/qmk_firmware #8022: No debug output detected by hid_listen or QMK Toolbox on lets_split

r4dr3fr4d, 2020-01-27:
Hacking lets_split to add a device/feature and need debug output for that, but can't even get it working on the default build without any of my feature additions.

 - added `debug_enable=true;` inside of `keyboard_post_init_user()` in `default/keymap.c`
- changed `CONSOLE_ENABLE` to `yes` in `rules.mk`, also added `DEBUG_ENABLE`
- and then `prints()` in `process_record_user()`

Am I missing anything?

r4dr3fr4d, 2020-01-27:
Forgot to add:  `hid_listen` doesn't even detect the keyboard, which I suppose is the real problem - tried with `sudo` as well.

ridingqwerty, 2020-01-27:
What operating system are you using?

r4dr3fr4d, 2020-01-27:
Arch Linux, sorry

ridingqwerty, 2020-01-27:
Ok, just checking.  I have issues with debugging in Ubuntu 18.04 (keyboard response issues) but running `hid_listen` as root or with sudo resolves that.

Have you tried adding a `DEBUG` keycode?

r4dr3fr4d, 2020-01-27:
No, because I don't have an actual keyboard (just using lets_split as an example to hack) but I can hook those pins up tonight and see if that helps.

r4dr3fr4d, 2020-01-28:
No luck with `DEBUG`.  Didn't expect it to work either, since hid_listen is stuck on `Waiting for device:...`

DLWood1001, 2020-02-08:
I am seeing the same symptom. Setup my build with COMMAND_ENABLE and debug_enable = true (also setup a debug key). When I start hid_listen, it just pends at 'waiting for device'. If I run the "lsusb" command, I do not see the FF31:0074 usb device that hid_listen is looking for.

I am running Fedora 31 with selinux disabled; pretty vanilla install actually.

fauxpark, 2020-02-09:
@DLWood1001 `COMMAND_ENABLE` will not turn on the Console interface; what you want is `CONSOLE_ENABLE = yes`.

Additionally hid_listen doesn't look for a VID/PID of `FF31:0074` so you will not see it in lsusb. Those numbers are an HID usage page/usage pair: https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/hid-usages

closed 2020-06-08 by stale[bot]