qmk/qmk_firmware #8023: [Bug] Odd behavior with KC_SFTENT

drasbeck, 2020-01-27:
This bug report is just addressing some findings I've done with the KC_SFTENT, which leads me to think that it is not working entirely as intended. I have been able to mitigate my own issues, and since I don't find more posts about it, we're probably not looking at a huge problem. But anyways:

I have set my Preonic Rev3 up with a KC_SFTENT key. In most situations it behaves just as intended (tap for `enter`, hold for `right shift`). But in a few pieces of software I get some strange behavior. The two programs I have found to be affected are Red Dead Redemption 2 on Windows 10 and [Alfred](https://www.alfredapp.com/) (the Spotlight replacement) on macOS Catalina.

In **Red Dead Redemption 2**, the enter functionality is sparse. It often coincides with the fact that the game wants me to press and hold `enter`, for some functionality, which of course doesn't work with KC_SFTENT. But in some situations it doesn't register a regular tap for `enter` as it should. _Mostly mitigated by reassigning the key, within the game._

In **Alfred**, whenever I hover over a file or app (or something that can show a preview) in Alfred and tap for `enter`. It yields the functionality of me tapping shift, which results in showing the file/app preview instead of the intended opening/executing that the press of enter should result in. _Mitigated by disabling the preview functionality of Alfred._

If there's any debug information I can provide, please let me know.

## System Information

 - Keyboard:  Preonic
   - Revision:  3
 - Operating system:  Windows 10 (latest stable build), macOS Catalina (latest stable)
 - Firmware built with QMK configurator on Dec 12, 2019, see [my repo](https://github.com/drasbeck/qmk-preonic-danish).

drashna, 2020-02-08:
Try using `LSFT_T(KC_ENTER)` instead, and see if that works better?

drasbeck, 2020-08-14:
I found (and disabled) a setting in Alfred, that shows a preview on pressing shift. This has absolved my problems with this behaviour all together. But I still think there is some sneaky little bug somewhere in the code for `KC_SFTENT`.

@drashna Thanks for your suggestion, I didn't test it yet, but it's a great idea, and should I happen to run into other software that shows problems with `KC_SFTENT` I'll definitely see if using `LSFT_T(KC_ENTER)` has any effect.

chengluyu, 2021-04-20:
> Try using `LSFT_T(KC_ENTER)` instead, and see if that works better?

Thank you. It works.

I’m using a Planck. I set right shift to `KC_SFTENT` and encountered the same problem as @drasbeck. I set the key to `LSFT_T(KC_ENTER)` and it works like a charm. Thanks, again.

malfario, 2021-07-16:
I'm having the same issue. Karabiner's Event Viewer (macOS) is showing a SHIFT press right before the actual ENTER key code is sent. That would explain why `KC_SFTENT` is triggering the app preview instead of launching it.

```
type:key_down        code:229        name:right_shift     misc:flags: right_shift 
type:key_up          code:229        name:right_shift     misc:
type:key_down        code:40         name:return_or_enter misc:
type:key_up          code:40         name:return_or_enter misc:
```