qmk/qmk_firmware #8028: Building firmware for a unequally split keyboard

imjustheretodomyjob, 2020-01-28:
> This is my first time building a keyboard, and I don't have much experience with programming.

I have built a hand-wired, split keyboard ( left side: 14 x 5, right side: 7 x 5) wit each side powered by a Arduino ProMicro. The halves are connected by a TRS cable, using serial communication. Now, I'm trying to build a firmware to be flashed on it.

[image]

About the keyboard:

The left side will be the master side. It is basically a 75% keyboard, so it has most of the keys, except the function row has been removed, and "Esc" has replaced the tilde key. The function row (F1 to F12) is in a secondary layer, under the top number row. The right side is the printscreen keys, the arrows, the numpad, and the tilde key (and an = key).

What I have done:

After referring to the QMK split keyboard guide, the nicinabox & u/campasachamp guides, as well as the olkb and mechanicalkeyboards subreddits, I was told that I could use the "Let's Split" firmware as a base, and edit that to suit my needs. So: 

1. I downloaded the source code for Let's Split
2. Edited the config.h file with the number of rows (which has to be doubled, according to the guide) and columns (I was told to just enter the higher number of columns). And I added the mapping to the pins on the microcontroller. I used the same layout for both, so they only differ for the columns.
3. Edited the rev2.h file with my layout (I think I was supposed to display the keyboard as an array, with the left and right sides mapped accordingly). The last line in this file defines a layout (as ortho_4x12). Should I change this name ?
4. I have to change the keymap.c file to match my setup. This is the part that I have gotten stuck at. I tried using kbfirmware.com to build my keymap, but it seems so simple compared to the let's split one. That has a whole visual array, and multiple layers. Do I just remove all of that and replace it with mine ? Do I try to swap those values with mine (like a retrofit) ?

What do I do now ? What should my next steps be ?

drashna, 2020-02-08:
Yeah, you should be able to do so.

https://docs.qmk.fm/#/feature_split_keyboard?id=hardware-configuration-options

imjustheretodomyjob, 2020-02-08:
> Yeah, you should be able to do so.
> 
> https://docs.qmk.fm/#/feature_split_keyboard?id=hardware-configuration-options

I tried following that guide. When I flash it to my keyboard, it says that it's done installing, and then does not detect any of the keys, only 1,Q,A,Z on the left side, and no other keys. I really don't know what is going wrong here.

closed 2020-06-07 by stale[bot]