qmk/qmk_firmware #8060: Failed to upload Redox Firmware from Linux

MarkuBu, 2020-02-01:
Hi,
I want to upload a firmware from Linux to the Redox Keyboard, but the upload failed and now the keyboard is dead.

I had to run make <my_keyboard>:<my_keymap>:flash with sudo to make the serial port available, but I just got error messages like this

```
avrdude: Error: butterfly programmer uses avr_write_page() but does not
provide a cmd() method.
 *** page 11 (addresses 0x590c - 0x598b) failed to write

Writing | ################################################## | 100% 0.32s

avrdude: 22924 bytes of flash written
avrdude: verifying flash memory against .build/redox_rev1_default.hex:
avrdude: load data flash data from input file .build/redox_rev1_default.hex:
avrdude: input file .build/redox_rev1_default.hex auto detected as Intel Hex
avrdude: input file .build/redox_rev1_default.hex contains 22924 bytes
avrdude: reading on-chip flash data:

Reading |                                                    | 0% 0.00savrdude: error: programmer did not respond to command: set addr
avrdude: error: programmer did not respond to command: set addr
```
I also tried arduino and dfu instead of flash, but dfu doesn't work (as expected) and arduino was the same result as flash.

What's wrong? I hope someone can help me because I don't want to dump 200€ just because of a software issue

Regards
Markus

zvecr, 2020-02-01:
You can try `sudo systemctl disable --now ModemManager.service` or the caterina rules at <https://docs.qmk.fm/#/faq_build?id=linux-udev-rules>.

asgergb, 2020-02-01:
@zvecr, you're a life saver! I happened to run into the exact same problem just now on the redox_w, and disabling ModemManager worked.

MarkuBu, 2020-02-02:
@zvecr disable tha ModemManager helps for me too. Big big thanks from me. Now I can start modifying the default layout

closed 2020-02-02 by MarkuBu