qmk/qmk_firmware #8061: Keyboard sending completely wrong keypresses

snalgli, 2020-02-01:
I have succesfully built and flashed a handwired GK64 Layout keyboard. It worked very fine for about half an hour, then I went to sleep that evening. 

On the next morning, when starting up my PC, the keyboard sent completely random keypresses. Some do nothing at all, some press key combinations like CTRL+N, or some combinations I didn't even know (I learned some new Windows features through this...) 

It shows the same behaviour when flashing again, also with different firmware it does the same thing. I don't know at all what could be the problem, since I didn't change anything at all.. Also it's bluetooth capable, but the same thing happens. 

I'm running an Adafruit Feather 32u4 Bluefruit LE, programming happened on Windows 10. Test objects are said PC with Win and an iPad.

Thanks for help in advance..

drashna, 2020-02-08:
Sounds like it may not be wired up correctly?   Or that the matrix layout is wrong.

snalgli, 2020-02-10:
It turned out I had to reset the EEPROM.. don't know what went wrong there

closed 2020-02-10 by snalgli

maberalc, 2020-09-14:
Hi, I got same problem here, could you tell me how did you reset the EEPROM? I'm using Bluefruit Feather 32u4, which is a Pro Micro with same bluetooth chip.

maberalc, 2020-09-14:
Keyboard is working properly wired but sends wrong keys by Bluetooth. BT is obviously well connected as it is soldered on same board as the mcu and it worked on a test with another keymap. Now when i tested with the real keymap it sends wrong keys. Do you know what could be the problem here?

maberalc, 2020-09-17:
How did you solve this exactly?

snalgli, 2020-09-17:
I use qmk Toolbox for flashing, there's a button at the top for resetting the eeprom. this solved it for me

Ciaanh, 2022-06-23:
Hi, I got the same issue today with an ELite-C and clearing the EEPROM solved it.