qmk/qmk_firmware #8069: Creating a dead/diacritica key

MarkuBu, 2020-02-02:
Hi,
I want to create a new minimalistic keyboard for a german layout called PUQ. Originally it uses a diacritica key to type the german characters ä, ö, ü and ß.

http://adnw.de/index.php?n=Main.OptimierungF%c3%bcrDieGeradeTastaturMitDaumen-Shift

First you type the base character like the a and then you hit the diacritica key and get the ä. If you type the A and the diacritica key you get the Ä and so on. But this is done in the driver on the PC. I want to be able to use my keyboard on any PC without the need for a driver. 

How can I create a similar behavior with QMK? 

Regards
Markus

drashna, 2020-02-08:
It sounds like a lot of that is handled by the OS's layout.

However, some of this can be done with the unicode feature. 
https://docs.qmk.fm/#/feature_unicode

MarkuBu, 2020-02-08:
yes. I think most of this is done in the os driver. I have to learn more about QMK, thanks.

currently I solved it with an OSL

closed 2020-05-09 by MarkuBu