qmk/qmk_firmware #8090: [Bug?] OSL + LCAG(kc) not working the way I think it should work

macintacos, 2020-02-05:
## Describe the Bug

I can't determine whether or not this is "intended" behavior, so I'm filing it as a bug just in case.

I have an OSL key to bring me to a certain layer. I want to be able to do the following:

1. Tap/release the OSL key. Now I'm in layer X momentarily. Next steps takes place without holding the OSL key.
2. In layer X, there is a key that I've specified with `LCAG(kc)` (for the purposes of this explanation, the `kc` doesn't matter).
3. I would like to hold the `LCAG` key and hit another key in that layer to trigger that modifiers + `kc` combo.
	1. For context, the `kc` that I'm hitting is not the one assigned to `LCAG(kc)`, it's a completely separate key.
	2. The `kc` in question that I'm hitting is different between the "base" layer (where you began) and layer X; for example, on the "base" layer it's assigned `Q` whereas on layer X it's assigned `1`.

The above doesn't work; once I hold the `LCAG` key, I'm popped back to the base layer. I tested with "vanilla" modifiers in layer X, and I can chain them all together just fine (holding control+alt+cmd+`kc` works as expected _in layer X_, and pops me back to the base layer after I've completed the incantation). As far as I can tell, there's no way to get the `LCAG` behavior behind a single key in the destination layer for OSL (and have it behave the same way as the vanilla modifiers work in that layer).

Hopefully I'm making sense; I can provide more clarification if the above isn't enough.

## System Information

 - Keyboard: Planck 
   - Revision (if applicable):  Rev 6
 - Operating system: macOS 10.15.2
 - (not including information about gcc/git since I'm compiling this via config.qmk.fm) 
 - Any keyboard related software installed? 
   - [x] Karabiner
   - [x] Other: Keyboard Maestro

drashna, 2020-02-08:
I think this is an interaction with the two features.  I've seen similar in the past, but I think it was fixed.  However, that appears to not be the case.

macintacos, 2020-02-17:
@drashna Yeah definitely doesn’t work and I can’t figure out a way to get around it. Worth it to try fixing again?

drashna, 2022-08-14:
Was this still an issue?