qmk/qmk_firmware #8106: [Bug] Inconsistent MATRIX_COLS and MATRIX_COL_PINS. (KB: facew)

elfmimi, 2020-02-06:
## Describe the Bug

I was checking consistency of MATRIX_ROWS and MATRIX_COLS regarding the actual number of elements in MATRIX_ROW_PINS and MATRIX_COL_PINS respectively, for all keyboards.
And found out this in **keyboards/facew/config.h**.

```
#define MATRIX_COLS 11
#define MATRIX_COL_PINS { A0, A1, A2, A3, A4, A5, A6, A7, C7, C6}
```

I suppose above should have been written like this.
`#define MATRIX_COL_PINS { A0, A1, A2, A3, A4, A5, A6, A7, C7, C6 C5 }`

Can anyone provide certain information on this issue?
@mechmerlin 

## System Information

 - Keyboard:  facew
 - QMK Firmware version: latest (0.7.140)

## Additional Context

https://github.com/qmk/qmk_firmware/blob/6a0e155afe607c75bfffe3f10fb73c9337942f52/keyboards/facew/config.h#L30-L34

fauxpark, 2020-02-06:
Good catch. The entire first column of the layout macros is comprised of `KC_NO`s, so that can be eliminated, `MATRIX_ROWS` set to 10, and `MATRIX_COL_PINS` either left as is, or set to `{ A1, A2, A3, A4, A5, A6, A7, C7, C6, C5 }` depending on which column that was meant to be.

elfmimi, 2020-02-06:
Digging a bit into history, there can be found a clue. Before this commit it once was a custom matrix.
https://github.com/qmk/qmk_firmware/commit/0231caa25aa258a79a7fb73cc57723655055b928#diff-e7904ba5a8e04def4cd10c689415cfbaL69-L78

elfmimi, 2020-02-14:
I guess this keyboard's circuit must be almost identical to schematics kept here.
https://github.com/showjean/ps2avrU/blob/master/firmware/ps2avrGB/Circuit/ps2avrGB/20140829_140848_001.png

elfmimi, 2020-02-19:
Attached here is a list of keyboards which seems to share similar design which originates to b.face .
[alike-of-b.face.txt](https://github.com/qmk/qmk_firmware/files/4226119/alike-of-b.face.txt)

Closest to facew among those is bfake and the following is the diff between the two.

```
---bfake/config.h      2020-02-11 21:38:52.972852841 +0900
+++ facew/config.h      2020-02-19 16:21:03.262556524 +0900
@@ -26 +26 @@
-#define PRODUCT         b.fake
+#define PRODUCT         facew
@@ -34 +34 @@
-#define MATRIX_COL_PINS { A0, A1, A2, A3, A4, A5, A6, A7, C7, C6, C5 }
+#define MATRIX_COL_PINS { A0, A1, A2, A3, A4, A5, A6, A7, C7, C6}
@@ -44,2 +43,0 @@
-
-#define NO_UART 1
```

```
--- bfake/bfake.h       2018-10-23 02:04:13.237323982 +0900
+++ facew/facew.h       2020-02-19 16:21:03.262556524 +0900
@@ -18,2 +18 @@
-#ifndef BFAKE_H
-#define BFAKE_H
+#pragma once
@@ -26 +25 @@
-    K12, K21, K22, K23, K24, K34, K35, K25, K26, K27, K28, K38,   K40,      \
+    K12, K21, K22, K23, K24, K34, K35, K25, K26, K27, K28, K38, K48,  K40,  \
@@ -34 +33 @@
-  { KC_NO, K41,   K42,   K43,   K44, K45, K46,   K47,   KC_NO, K49,   K40},   \
+  { KC_NO, K41,   K42,   K43,   K44, K45, K46,   K47,   K48,   K49,   K40},   \
@@ -56,2 +54,0 @@
-
-#endif
```

note: K48 is the key next to right shift which is used to support ISO layout.

closed 2020-12-04 by elfmimi