qmk/qmk_firmware #8131: [Survey] Let us know what OS you use!

Erovia, 2020-02-09:
Hi guys!

Sometimes moving forward means leaving the old behind.
As resources are limited, it's important for every project to know their user-base.
Right now, we'd like to know which OS do you use when you compile/flash/hack QMK.

We'd be very grateful if you could fill out our form.
It's just one simple question, the survey is anonymous.

https://forms.gle/djrMfZWuVS7oZerj9

If the shortened link does not work, you can try this:
https://docs.google.com/forms/d/e/1FAIpQLSe-CjwjqKo2yIFkXCPDrpRgmBStpaS7c8nt7UIlSw59amzjiQ/viewform

Thank you and happy hacking and clacking!

chung1905, 2020-02-18:
Will you public the result?

Erovia, 2020-02-18:
Yes, I'm planning to publish the results in some form.

closed 2020-02-19 by Erovia