qmk/qmk_firmware #8193: [Bug] Bootmagic lite trigering with sym_g debounce on ATMEGA328p

yiancar, 2020-02-18:
## Describe the Bug

When running on an atmega328p with default debounce, bootmagic lite cannot trigger.
I have checked the timers and the setting should be the same as 32u4. Timer0 (8bit) is being used.

I cannot seem to figure out why or how the debouncing is not allowing for the value to propagate through. Every time I read the row at the initial stage, no columns are detected even when in reality they are. This is not the case in 32u4. 

I wonder if somehow 328 is slower or something weird like that.

With both eager debounce method this problem is not observed.

@alex-ong If you could take a look I would really appreciate it. 328p doesnt have console output but I use a few leds to debug. If you are unaware of what bootmagic lite does, basically right after `matrix_init`, it issues 2 `matrix_scan()` with a delay of 2*debounce in between and then checks if a key is pressed using: `(matrix_get_row(row) & ((matrix_row_t)1<< col)
`

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

alex-ong, 2020-02-18:
Hi, there was a precious bug where the timer wasn't initialized at bootloader stage and hence the call to

timer_read() would always return 0.

Have you verified that 
a) sym_g works on 32u4
B) that timer is initialized on 328p?

> Timer0 (8bit) is being used.

Was it turned?

Related thread:

(To link)

fauxpark, 2020-02-18:
`sym_g` is the default debounce algo, so it definitely works on 32U4.

Timer0 is the one that drives `timer_read()`, `timer_elapsed()` etc. so if that isn't running, a whole bunch of other stuff will be broken too, I think. The problem, as discussed in the linked issue ^ is that interrupts are not enabled yet when Bootmagic Lite is triggered, so the timer does not start ticking.

alex-ong, 2020-02-18:
Sorry, my bad for spreading misinformation. The interrupts were the problem (and thereby calls to timer_elapsed()) would give wrong values

Are you sure it works on 32u4 @fauxpark? The linked issue makes it seem that bootmagic wouldn't work on any keyboard using 32u4 and sym_g...

fauxpark, 2020-02-18:
Yup, I just tested with my Satan. This only affects boards that use V-USB (so, only the 32A and 328P right now) because `sei()` is not called before `keyboard_init()` (which immediately calls `timer_init()`, then `matrix_init()`, which leads to `bootmagic_lite()`).

yiancar, 2020-02-22:
aaand fixed:)

closed 2020-02-22 by yiancar