qmk/qmk_firmware #8197: [Feature Request]

Supermagnum, 2020-02-18:
## Feature Request Type

- x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
Please add support for the European signs Ø Æ Å Ä Ö þ ð.
Also, please add support for Norwegian colemak and Norwegian Dvorak layout.
Norwegian Dvorak:
https://github.com/Thhethssmuz/Norwegian-Dvorak
https://dvorak.tg90nor.net/

Norwegian colemak and variants can be found here:
https://forum.colemak.com/topic/2315-dreymars-big-bag-of-keyboard-tricks-main-topic/

noroadsleft, 2020-02-18:
Support for Colemak and Dvorak is already included.

- https://github.com/qmk/qmk_firmware/tree/1566f0a9978b7799741d9499228d525aa5019258/quantum/keymap_extras/keymap_colemak.h
- https://github.com/qmk/qmk_firmware/tree/1566f0a9978b7799741d9499228d525aa5019258/quantum/keymap_extras/keymap_dvorak.h

The Norwegian part of your request may better served on the side of the computer's operating system.

QMK doesn't send characters to the host system, it sends scancodes, and it's up to the host's OS what happens when a particular scancode is received. A scancode message is more like "My user pressed the second key on the second row," rather than "My user pressed the Å key." All keyboard firmwares work this way, from a $5 rubber dome to a $1,000 boutique custom mechanical keyboard.

Erovia, 2020-03-12:
Closing this issue as there is nothing else left to add.
If you still have related questions, feel free to reopen this case.

closed 2020-03-12 by Erovia