qmk/qmk_firmware #8202: doc issue

ymzcdg, 2020-02-19:
https://github.com/qmk/qmk_firmware/blame/master/docs/getting_started_introduction.md#L7
I don't know if ‘keyboard’ folder exists

noroadsleft, 2020-02-19:
Ahh, good spot.

The `keyboards` directory was originally named `keyboard`. It should read `keyboards`.

https://github.com/qmk/qmk_firmware/pull/432

ymzcdg, 2020-02-20:
> 
> 
> Ahh, good spot.
> 
> The `keyboards` directory was originally named `keyboard`. It should read `keyboards`.
> 
> #432

Thanks.
Fix this in https://github.com/qmk/qmk_firmware/pull/8185
Please merge.

ymzcdg, 2020-03-13:
https://github.com/qmk/qmk_firmware/blame/master/docs/understanding_qmk.md#L133
How can I understand the word"cluecard"?I can't find this word in my dictionary.

noroadsleft, 2020-03-14:
> 1.https://github.com/qmk/qmk_firmware/blame/master/docs/understanding_qmk.md#L133
> How can I understand the word"cluecard"?I can't find this word in my dictioary.

`cluecard` is an old device by skullydazed, and refers to [clueboard/card](https://github.com/qmk/qmk_firmware/tree/dc79792ab463f614e3aba979dace7d9625b340bf/keyboards/clueboard/card). `cluecard` should probably be capitalized here (`Cluecard`).

> 2.https://github.com/qmk/qmk_firmware/blame/master/docs/understanding_qmk.md#L131
> is the"process"a noun or a verb？

Verb.

ymzcdg, 2020-03-14:
> `cluecard` is an old device by skullydazed, and refers to [clueboard/card](https://github.com/qmk/qmk_firmware/tree/dc79792ab463f614e3aba979dace7d9625b340bf/keyboards/clueboard/card). `cluecard` [...]

If I use Cluecard, I will find keyboard/keymap level functions. Is that right?

noroadsleft, 2020-03-14:
> If I use Cluecard, I will find keyboard/keymap level functions. Is that right?

Yes. It's used as a sample implementation.

closed 2020-07-13 by stale[bot]