qmk/qmk_firmware #8203: redox wireless, D4 on receiver

danielo515, 2020-02-19:
Hello,
I noticed that the redox wireless has a fourth led on the receiver labeled D4.
However, only three of the leds seems to be controllable from the firmware. Is there any reason for this?
Is it possible to control the fourth led ? It will be a real advantage!

noroadsleft, 2020-02-19:
@mattdibi Could you please advise?

mattdibi, 2020-02-20:
Hi,
actually yes it's perfectly usable. [Here](https://github.com/mattdibi/redox-keyboard/issues/24#issuecomment-500109781) I put a few references on how to use it. It's been some time since I wanted to put the 4th led code but never find the time to actually work on it, I'm sorry.

danielo515, 2020-02-22:
Thanks for the link, I'll take a look at it. What do you mean by the old api? How is this led different from the other three?

mattdibi, 2020-02-22:
> What do you mean by the old api?

Now there's a [higher level API](https://docs.qmk.fm/#/custom_quantum_functions?id=led-control) to control LEDs but is more readable and easier to use.

> How is this led different from the other three?

It is not. It's exactly the same and should be used the same way.

danielo515, 2020-02-23:
But is there code to use D4 right now? Like the existing defs to turn on the blue led, magenta color and such?

mattdibi, 2020-02-23:
> But is there code to use D4 right now? Like the existing defs to turn on the blue led, magenta color and such?

No, there isn't. You need to add the changes I implemented [here](https://github.com/mattdibi/qmk_firmware/commit/e0d24216bcaaed86a148730b6cb74b97d3685f1d) and add your own logic.

Hope this helps.

closed 2020-06-22 by stale[bot]