qmk/qmk_firmware #8210: Is still mandatory to leave transparent the key that activates a layer?

danielo515, 2020-02-20:
Today I made this question on reddit:

>I recently received my redox and while configuring it I noticed that I added some momentary layer toggle keys that were not transparent on the target layer. It took me a while to notice because I didn't had any problem. If I remember correctly you should got locked on the toggled layer because the modifier key is no longer a modifier key in the new layer. So has something changed? Or it was always like this? Honestly, it will be a big advantage because I usually have several keys to access the same layer, and having to make "holes" on the target layer makes it way less useful.

@drashna already answered me this:

> IIRC, this was the old behavior, that required the "prevent stuck modifiers" option to be enabled. The default behavior shouldn't require that

But the documentation still advice that you should make sure the target layer has a transparent key on the key that toggles that layer:
https://docs.qmk.fm/#/faq_keymap?id=modifierlayer-stuck

I suspect that documentation needs to be updated? Based on my experience I think @drashna is right and this is no longer necessary.

ridingqwerty, 2020-02-20:
I've come across users in other communities who likewise claimed that `KC_TRNS` was no longer needed to prevent layer lock, and was a little stunned at the time and couldn't be bothered to test.  Since reading this issue, I've tested with `MO` and `LT` and am rather surprised to learn this is no layer lock.  Color me intrigued.

That said, I'm hesitant to recommend updates to this documentation at this time for a few reasons:

- It doesn't really read to me as proscriptive, but rather that this can happen and here is how to address it.
- Plenty of users don't update their repo constantly, so they are very much still using `KC_TRNS` under layer movement key.

drashna, 2020-03-04:
No, it's not necessary, but it is still recommended.   

For instance, I helped somebody a couple of days ago when using TG wrong.  UJust using a transparent key on the destination layer would have prevented the problem, basically. 

There are a number of other cases where this may also be the case.   

But as I said, it's not necessary.  But unless you know what you're doing, it can cause confusing behavior.

closed 2020-07-02 by stale[bot]