qmk/qmk_firmware #8211: [Bug] THe problem is back (OSM shift)

danielo515, 2020-02-20:
Hello, I want to report an odd behavior that I get with one shot modifiers from time to time.
I am aware that this was a known problem before: https://github.com/tmk/tmk_keyboard/issues/67
Seems that the problem was fixed by using one shot modifiers, but my problem is precisely with one shot modifiers, so I'm not sure how related they are.

## Describe the Bug

When I use the shift key as a one shot modifier some times I get two keys shifted up instead of two. It usually happens when I type fast, like when you type usual words such as ME.

## System Information

 - Keyboard:  ergodox and redox_w
 - Operating system: OSX (whatever is latest)
 - AVR GCC version: whatever is inside the official docker image
 - ARM GCC version: whatever is inside the official docker image
 - QMK Firmware version: 0.7.120
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [x] Karabiner
   - [ ] Other:

## Additional Context

If it helps, here is my config.h (this is the simplest one on a keyboard where the problem exists)

```
#define IGNORE_MOD_TAP_INTERRUPT
#define TAPPING_TERM 200
```
and my rules.mk

```
TAP_DANCE_ENABLE = yes # Enable the tap dance feature.
COMBO_ENABLE = yes
LEADER_ENABLE = yes
CONSOLE_ENABLE = no
LTO_ENABLE = yes # link time optimizations
```

I also have enabled combos, tap dance, leader key

drashna, 2020-03-04:
Could you try disabling/uninstalling Karabiner and see if the issue is fixed.  It could be an interaction between the two things that is causing this issue.

danielo515, 2020-03-05:
Sure, will try

kabuko, 2020-08-05:
For me, this was an issue specifically with Karabiner and this was also the case [outlined in a previous issue](https://github.com/qmk/qmk_firmware/issues/6571). I would love to hear a solution other than disabling Karabiner, but that's probably an issue to bring up with Karabiner rather than QMK.

danielo515, 2020-08-06:
I have to say that, since I have Karabiner disabled I don't have this problem anymore.

closed 2022-08-28 by zvecr