qmk/qmk_firmware #8212: [Help] Triggering unwanted combos

danielo515, 2020-02-20:
Hello.
I can't say this is a bug of the combo feature, because in some way it is working as expected, but not as I expect.
I have several combos defined for keys that I use often, and I use them often one after the other. For example `ui` is a combo that should send esc, but it is also an usual combination on keys like Gui.
To avoid unwanted combo trigger I have a very very low combo time configured, as low as 40ms, but the combo is still triggered. I don't think I'm hitting those keys that fast, so probably there is an amount of time you can no go lower.

I would take any advice to improve this situation.
It is important to say that it happens more often in some keyboards than others. For example, my ergodox does this more often (gold clicky switches) than my redox (silent reds). Maybe the switch type makes a difference? Is my Ergodox too slow and it is not able to go that low in terms of timing?

Thanks and regards

danielo515, 2020-03-19:
Sorry for bumping but, any suggestion? Thanks!

cari66ean, 2020-04-21:
I would try going even even lower with your combo time. I can still activate them easily with a timing of 20 or 25ms but it makes accidental presses much less likely.

danielo515, 2020-04-22:
Ok, I'll try that. Honestly, I thought that below certain amount it is impossible for the atmega, but I'll try 20 or 25

cari66ean, 2020-04-22:
> 
> 
> Ok, I'll try that. Honestly, I thought that below certain amount it is impossible for the atmega, but I'll try 20 or 25

It might depend a little from controller to controller. However with 40-50ms when I tried typing keys one after another but still rather quickly like 'u' and 'i' when there's also a 'ui' combo, I would trigger the combo almost exclusively too.

danielo515, 2020-04-22:
I just lowered it to 25 and seems that it works much better! I can now type ui without problems (almost). Honestly, I didn't thought that I can type two keys that fast (lower than 40ms), but seems that I was doing that...
I'll try for a couple of days and re-open the issue if it comes back. Thank you!

closed 2020-04-22 by danielo515