qmk/qmk_firmware #8241: [Bug] Nix environment cannot build Massdrop firmware

emptyflask, 2020-02-24:
Probably other keyboards as well, but I haven't tested. I'm currently running `nix-shell --pure --run "make"` which will take a very long time to finish...

I'm using Nix on MacOS. I've tested this on both `master` and `future` branches.
Building without Nix works fine. I haven't tried this on Linux/NixOS yet.

## Describe the Bug

```
❯ make clean; nix-shell --pure --run "make massdrop/alt:default"

QMK Firmware 0.7.168
Deleting .build/ ... done.
QMK Firmware 0.7.168
Making massdrop/alt with keymap default

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/massdrop/alt/matrix.c                                                          [OK]
Compiling: keyboards/massdrop/alt/config_led.c                                                      [OK]
Compiling: keyboards/massdrop/alt/keymaps/default/keymap.c                                          [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: tmk_core/common/arm_atsam/eeprom.c                                                      tmk_core/common/arm_atsam/eeprom.c: In function 'eeprom_read_byte':
tmk_core/common/arm_atsam/eeprom.c:27:24: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
     uintptr_t offset = (uintptr_t)addr;
                        ^
tmk_core/common/arm_atsam/eeprom.c: In function 'eeprom_write_byte':
tmk_core/common/arm_atsam/eeprom.c:32:24: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
     uintptr_t offset = (uintptr_t)addr;
                        ^
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_massdrop_alt_default/common/arm_atsam/eeprom.o] Error 1
make: *** [Makefile:579: massdrop/alt:default] Error 1
Make finished with errors
```

## System Information

 - Keyboard:  Massdrop ALT High-Profile
 - Operating system: MacOS 10.14.6
 - AVR GCC version: 8.1.0
 - ARM GCC version: 6.3.1
 - QMK Firmware version: 0.7.162

## Additional Context

I briefly attempted to update `shell.nix` to a more recent version of nixpkgs, but the situation there is even worse:
```Package ‘avr-libc-2.0.0’ in /nix/store/xxxxx is not supported on ‘x86_64-darwin’, refusing to evaluate.```

emptyflask, 2020-02-26:
I got it to build under NixOS by running:
```nix-shell --pure --arg avr false --run "make massdrop/alt:default"```
(the --pure flag is optional, but probably good to use anyway)

drashna, 2020-03-04:
This may be related to the ARM GCC version, but I can't reproduce this locally, checking out the tag/version that you've mentioned. 

Also, it may be worth re-running the qmk_install.sh script.

emptyflask, 2020-03-04:
@drashna I get the same under NixOS (which simply can't use the install script -- in this environment, the shell.nix is intended to replace that)

The problem is that the AVR flags are included by default, and conflict with the ARM compilation.
```nix
{ avr ? true, arm ? true, teensy ? true }:
# ... lots of omitted stuff
{
  CFLAGS = lib.optional avr avr_incflags;
  ASFLAGS = lib.optional avr avr_incflags;
}
```

If `avr` is overridden on the command line using `nix-shell --arg avr false`, it works fine.

tzarc, 2020-03-04:
Not familiar with nix; but it seems like it's unconditionally using the AVR compiler flags with gcc, regardless of whether it's an Arm build or not?

Guess I need to install nix and find out...

tzarc, 2020-03-04:
Yep, confirmed repro case locally.
Unsure how to solve this to be honest -- as far as I can tell we don't really have a way to determine at the nix level whether we're building an AVR or an Arm firmware.
Will have to have a bit of a think, perhaps we can do something like stick them into a different variable through nix, then if the normal make system detects an AVR chip we append the new variable to CFLAGS etc.

tzarc, 2020-03-04:
Yeah, that looks like it did the trick. Test build going on locally, will PR it if we're all green.

emptyflask, 2020-03-04:
Cool. That sounds like what I was just thinking about in #nixos in IRC.

tzarc, 2020-03-10:
Can we get a confirmation that it's all good now?

emptyflask, 2020-03-10:
Yeah it works for me!

closed 2020-03-10 by tzarc