qmk/qmk_firmware #8242: [Bug][Keyboard] Ergodox EZ issues with debouncing and scan rate

drashna, 2020-02-25:
## Describe the Bug

Since #8092, the matrix scan and debouncing has been a problem. 

I've had to up the debounce from 30 to 60, so I don't get repeated key presses.  However, now, if feels like the debouncing is causing it to be less "smooth" (per @ezuk)

Also, the ergodox ez uses the "eager per row" debouncing, and ... it should probably stay on this debounce type, because of how the matrix is set up. 

Tagging @alex-ong, since they wrote the debounce API. 

## System Information

 - Keyboard:  
   - Revision (if applicable):  EZ
 - Operating system: Win, Mac, Linux
 - AVR GCC version: 8.3.x
 - QMK Firmware version: 0.7.166
 - Any keyboard related software installed: No

(un/related, it may be nice to handle the main matrix and the IO expander matrix debouncing separately, but that would mean using the custom debounce option, or rewriting the debounce API to support that)

alex-ong, 2020-02-25:
Hi,

For reasons unknown to me, the Ergodox EZ requires <x>ms~ debounce. Back when the implementation was cycle based, it was "5 milliseconds" but really meant 5 cycles at y hz scanrate, which was x ms. It used a symmetric global timer afaik.

At some point someone smarter than I wrote a custom eager debounce per row algorithm, pretty much directly for the Ergodox EZ. This was also cycle based , and kept the 15ms debounce.

I wrote an eager_pr algorithm, which is now the de facto debounce for ergodox ez. It used Milliseconds. People complained that it was buggy, and i did a test with the previous method (not my method), and my own method. By calculating scanrate i found that the millseconds required in my method to match the scanrate of the previous method was <x>. When people tested both methods with the "equivalent" debounce, they found both worked equivalently.

Not sure why milliseconds has increased from 30 to 60 for debounce; it should be the same....

drashna, 2020-02-25:
I suspect it was @seebs or @jackhumbert that wrote that code, but I could be wrong. 

But there was a recent PR that improved the scan rate by removing some apparently unnecessary i2c commands (as per spec sheets), and this lead to the issue. 

So, if eager_pr is based on/tuned for scan rate, then it would be broken by this improvement.

seebs, 2020-02-25:
I have one that I haven't had time to rewrite/merge as a debounce algorithm that worked excellently for me with the ergodox EZ, the core insight of which is that bouncing can't happen on a switch that hasn't been pressed recently-or-just-now. So, if a switch is considered down, the first flicker of anything is accepted as a key down, but it requires multiple consecutive scans of key release to report a key release, and this seems to work quite nicely. It wasn't really tuned by scan rate except in that you wanted enough cycles to clear the 5ms or so of key debounce... which is actually potentially an issue, I guess?

alex-ong, 2020-02-25:
@seebs Right, that's an eager on key-down, with a non-eager x scan release. I'm talking about the commit that was actually in master; not sure if it was actually yours. I am definitely familiar with your work on the issue.

@drashna eager_pr is not tuned for a specific scan rate. It should work regardless of scan rate. We could always implement the original baseline (roughly a year ago) algorithm, since I think that's what people were originally happy with. AFAIK it is essentially eager_pr but based on num_scans rather than milliseconds.

Also note that I don't have an ergodox_ez, but I am happy to implement the 1 year ago~ algorithm for people to test. I believe that its the same algorithm as eager_pr (though written in a completely different way).

alex-ong, 2020-02-25:
Hi, i'm linking the original thread (#6269)

At this point I'm not certain the the issue is entirely debounce related.

If you want to use the "reference working" debounce located at commit
cd544e10a,

Then look at this commit:
https://github.com/qmk/qmk_firmware/compare/master...alex-ong:ergoez_debounce

Should be just two files: `debounce.c`, then changing rules.mk to use debounce.c rather than eager_pr.

If that doesn't work i'd tentatively rule out the debounce algorithm.

ggayan, 2020-03-31:
Most likely relevant here: https://github.com/qmk/qmk_firmware/pull/8587

I'm testing the PR in my Ergodox EZ with debounce=5 and eager_pr and so far so good

drashna, 2020-08-01:
The wrapping timer PR for debouncing seems to have fixed this.  Closing.

closed 2020-08-01 by drashna