qmk/qmk_firmware #8259: [Bug] [matrix/noah] Mod-Tap requires 2 taps to initiate

fobbyal, 2020-02-27:
## Describe the Bug
I have configured `LCTL_T(KC_ESC)` for the caps-lock position and and it requires a double tap to trigger escape initially. Actually if i started tapping fast continuously the first 2 taps register as one escape but the following taps registers as expected. For example, 
```
Tap -> 150ms -> Tap -> 150ms -> Tap -> 150ms -> Tap -> 150ms -> Tap -> 150ms -> 
```
will only register 4 escape not 5 escapes. If i just Tap once it doesn't register anything. It seems that the driver only recognizes the Tab if it is somehow triggered in a very short amount of time. I have other keyboards with the same Mod-Tap configuration registered but do not have the issue. We may need some help form the maintainer @yulei . Here is my fork that re-produces this issue. I would like too help but i only have limited experience in C and don't know where to start.

mod tab requires 2 taps to initiate. 

## System Information

 - Keyboard:  matrix/noah
   - Revision (if applicable):  
 - Operating system: OSX
 - AVR GCC version: 8.3.0

 - ARM GCC version:  8.3.1

 - QMK Firmware version: 0.7.109

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [x] Karabiner (doesn't work even if i shut it down)
   - [ ] Other:

## Additional Context

yulei, 2020-03-02:
Maybe the following lines in the config.h should be removed. they were used as a work around before the fix by #7595 .
```
// tapping setting
#define TAPPING_TERM 200
#define RETRO_TAPPING
#define PERMISSIVE_HOLD
```

fobbyal, 2020-03-03:
I’ll try it out and report back.

fobbyal, 2020-03-03:
it didn't work with those lines commented out. I am going to try to debug a little when i have some free time. You have any idea where i can print out some debug statements off the top of your head?
[image]

drashna, 2020-03-04:
One thing to note, that should be `DEBOUNCE`, not `DEBOUNCING_DELAY`. I don't think that this would cause this, but is something that should be fixed anyways.

yulei, 2020-03-04:
it's for the custom debouncing matrix code, I'll try to use the standard matrix to check  if this could help.

yulei, 2020-03-05:
final got sometime to check this issue, I use segger rtt view to print debug message:
tag once
```
r/c 01234567
00: 000000000000000
10: 000000000000000
20: 100000000000000
30: 000000000000000
40: 000000000000000

---- action_exec: start -----
EVENT: 0200d(50277)Tapping: Start(Press tap key).
TAPPING_KEY=Record: row:2, col:0, pressed:d, time:50277, int:n, count:0

processed: 0200d(50277):0

r/c 01234567
00: 000000000000000
10: 000000000000000
20: 000000000000000
30: 000000000000000
40: 000000000000000

---- action_exec: start -----
EVENT: 0200u(50367)Tapping: First tap(0->1).
TAPPING_KEY=Record: row:2, col:0, pressed:d, time:50277, int:n, count:1

ACTION: ACT_LMODS_TAP[1:29] layer_state: 00000000(0) default_layer_state: 00000000(0)MODS_TAP: Tap: register_code

resiger_code:41
send keyboard report
in send_keyboard()
usbStartTransmitI
keyboard_report: 00 00 29 00 00 00 00 00
waiting_buffer_enq: { [3]=Record: row:2, col:0, pressed:u, time:50367, int:n, count:1
 }
---- action_exec: process waiting_buffer -----
Tapping: Tap release(1)
ACTION: ACT_LMODS_TAP[1:29] layer_state: 00000000(0) default_layer_state: 00000000(0)MODS_TAP: Tap: unregister_code

unresiger_code:41
send keyboard report
in send_keyboard()
usbStartTransmitI
keyboard_report: 00 00 00 00 00 00 00 00
TAPPING_KEY=Record: row:2, col:0, pressed:u, time:50367, int:n, count:1

processed: waiting_buffer[3] = 0200u(50367):1

Tapping: End(Timeout after releasing last tap): FFFFu(50569)
TAPPING_KEY=Record: row:0, col:0, pressed:u, time:0, int:n, count:0
```
seems everything works fine, this report was sent by usbStartTransmitI(), but the PC did not received.

tag twice
```
r/c 01234567
00: 000000000000000
10: 000000000000000
20: 100000000000000
30: 000000000000000
40: 000000000000000

---- action_exec: start -----
EVENT: 0200d(24885)Tapping: Start(Press tap key).
TAPPING_KEY=Record: row:2, col:0, pressed:d, time:24885, int:n, count:0

processed: 0200d(24885):0

r/c 01234567
00: 000000000000000
10: 000000000000000
20: 000000000000000
30: 000000000000000
40: 000000000000000

---- action_exec: start -----
EVENT: 0200u(24975)Tapping: First tap(0->1).
TAPPING_KEY=Record: row:2, col:0, pressed:d, time:24885, int:n, count:1

ACTION: ACT_LMODS_TAP[1:29] layer_state: 00000000(0) default_layer_state: 00000000(0)MODS_TAP: Tap: register_code

resiger_code:41
send keyboard report
in send_keyboard()
usbStartTransmitI
keyboard_report: 00 00 29 00 00 00 00 00
waiting_buffer_enq: { [4]=Record: row:2, col:0, pressed:u, time:24975, int:n, count:1
 }
---- action_exec: process waiting_buffer -----
Tapping: Tap release(1)
ACTION: ACT_LMODS_TAP[1:29] layer_state: 00000000(0) default_layer_state: 00000000(0)MODS_TAP: Tap: unregister_code

unresiger_code:41
send keyboard report
in send_keyboard()
usbStartTransmitI
keyboard_report: 00 00 00 00 00 00 00 00
TAPPING_KEY=Record: row:2, col:0, pressed:u, time:24975, int:n, count:1

processed: waiting_buffer[4] = 0200u(24975):1

r/c 01234567
00: 000000000000000
10: 000000000000000
20: 100000000000000
30: 000000000000000
40: 000000000000000

---- action_exec: start -----
EVENT: 0200d(25069)Tapping: Tap press(2)
ACTION: ACT_LMODS_TAP[1:29] layer_state: 00000000(0) default_layer_state: 00000000(0)MODS_TAP: Tap: register_code

resiger_code:41
send keyboard report
in send_keyboard()
usbStartTransmitI
keyboard_report: 00 00 29 00 00 00 00 00
TAPPING_KEY=Record: row:2, col:0, pressed:d, time:25069, int:n, count:2

processed: 0200d(25069):2

r/c 01234567
00: 000000000000000
10: 000000000000000
20: 000000000000000
30: 000000000000000
40: 000000000000000

---- action_exec: start -----
EVENT: 0200u(25159)Tapping: Tap release(2)
ACTION: ACT_LMODS_TAP[1:29] layer_state: 00000000(0) default_layer_state: 00000000(0)MODS_TAP: Tap: unregister_code

unresiger_code:41
send keyboard report
in send_keyboard()
usbStartTransmitI
keyboard_report: 00 00 00 00 00 00 00 00
TAPPING_KEY=Record: row:2, col:0, pressed:u, time:25159, int:n, count:2

processed: 0200u(25159):2

Tapping: End(Timeout after releasing last tap): FFFFu(25361)
TAPPING_KEY=Record: row:0, col:0, pressed:u, time:0, int:n, count:0
```
the pc only received one ESC. so seems it was caused by the low level usb driver.
seems the latest QMK master was upgrade to chibios 19, unfortunately noah can't work, it was  blocked during the system clock setting. 

@drashna is there any working keyboard which use the STM32F4?

fobbyal, 2020-03-06:
@yulei thanks for looking into this. I just realized that there was something that might be important that i forgot to report. I am using the LT keycode for layer switching on the SPACE key. the exact config is `LT(2,KC_SPC)` and this key behaves as expected when tapped. I am not sure if somehow the caps-lock position is special or the implementation between `LT` and  `LCTL_T` is drastically different.

yulei, 2020-03-06:
@fobbyal, it should fixed at https://github.com/yulei/qmk_firmware/tree/noah_tap, could you have a try?
The bug disappeared after I upgrade the configuration files of noah keyboard according the chibios in the master.
  
@drashna I finally found that why I could not use the standard matrix code. The A9 pin was default to sense VBUS in STM32 device who used OTGv1 driver. Even I always turned the VBUS sense off, the A9 pin still does not work correctly in the standard matrix code.

fobbyal, 2020-03-06:
@yulei i'll give it a try and report back

fobbyal, 2020-03-06:
@yulei i confirmed that it is fixed.. it fixed the leds for me too. It was kind of behaving strangely before. Thank you so much for your effort. I know that is a big upgrade going on. With that in mind when do you think this will land in master?

yulei, 2020-03-07:
yes, I also fixed some leds issues after the boards were shipped. I'll make a PR asap. The A9 pad's issue was still not fixed, so I have to use custom matrix codes.

yulei, 2020-03-07:
@fobbyal you can also merge #7803 to your own branch, which emulates EEPROM with flash, thus you can save your setting permanent. And this codes were verified by hundred people.

fobbyal, 2020-03-07:
thank you again.

tzarc, 2020-03-07:
> The A9 pad's issue was still not fixed

I apologise if you've tried these before, as I haven't been following the context. Have you tried setting `#define BOARD_OTG_NOVBUSSENS TRUE` in config.h, and setting A9's alternate function to mode 0 in the board init routines?

tzarc, 2020-03-07:
Never mind; I see `BOARD_OTG_NOVBUSSENS` is defined. You've tried setting AF0?

yulei, 2020-03-07:
yes, I have test reset this PIN to AF0 after initialized the usb driver(the VBUSNOSENSE register bit was set at the usbStart() call, It should be released after this register setting according to ST's datasheet),  but it only works in OUTPUT mode, I'm not sure why it can't  work in INPUT mode. Lately I'll use nucleo board with ST's HAL for a testing.

PR #8333 (2020-03-07): fixed #8259

## Description
upgrade board configurations to the latest chibios and fixed  #8259

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

*  fixes #8259

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [x] I have read the [**CONTRIBUTING** document](https://docs.qmk.fm/# [... cut]
commit e0d15e61 (2020-03-07): fixed #8259 (#8333)
closed 2020-03-07 by zvecr