qmk/qmk_firmware #8276: [Discussion] Software to assist with ongoing translation efforts

skullydazed, 2020-03-01:
In the discussion for #8252 @shelaf requested that we setup some translation support tools. [Transifex](https://www.transifex.com/) was mentioned but it is cost prohibitive for a small project like QMK.

This issue will serve as a place to discus other tools that may fill the same need, so we can evaluate their suitability for QMK.

shelaf, 2020-03-01:
For your reference, Trasifex FAQ:
> What are the basic requirements to qualify as an Open Source account?
> 
> We are proud and humbled to be hosting the largest Open Source translation community in the world. We offer Transifex for free to Open Source projects that have no funding, revenue, and/or commercialization model. If you have a question whether your organization qualifies for our Open Source program, please contact us.

Erovia, 2020-03-12:
A self-hosted option: https://wiki.gnome.org/DamnedLies

closed 2022-08-28 by zvecr