qmk/qmk_firmware #8281: ‘uint16_tword’ undeclared

grassbl8d, 2020-03-01:
Hi,

I'm getting this error when copmiling lately.

quantum/quantum.c:646:5: error: ‘uint16_tword’ undeclared (first use in this function)
     uint16_tword = (number >> 16);
     ^
quantum/quantum.c:646:5: note: each undeclared identifier is reported only once for each function it appears in
quantum/quantum.c:647:15: error: ‘word’ undeclared (first use in this function)
     send_word(word);

I think this might be related to a recent change . Can't compile firmware right now am getting this error.

[image]

Thanks

maciejmatczak, 2020-03-01:
https://github.com/qmk/qmk_firmware/commit/1ec8a7205f1719496e043776edcc972125fc57e3

It seems that formatting without full CI let that issue through...

fauxpark, 2020-03-01:
Duplicate of https://github.com/qmk/qmk_firmware/pull/8280

closed 2020-03-01 by grassbl8d