qmk/qmk_firmware #8302: [Bug] QMK-DFU exit key and bootmagic bootloader key don't work on the same key

Maartenwut, 2020-03-03:
## Describe the Bug

By default, the key in the first row and first column is the key that triggers the bootloader jump if pressed when the usb cable is plugged in. However, if the QMK-DFU escape key is set using QMK_ESC_OUTPUT and QMK_ESC_INPUT, it works too well. You escape the bootloader, but then it detects the key being pressed and brings you right back to the bootloader. This is usually the esc key.

The expected behavior would be that it actually goes back to the user program.

## System Information

 - Keyboard:  any that uses qmk-dfu and bootmagic (lite)
   - Revision (if applicable):  
 - Operating system: Windows 10 Pro 1903
 - AVR GCC version: 4.9.2

 - ARM GCC version:  4.9.3

 - QMK Firmware version: 0.7.119 

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
I don't know if it's fixed already, I tried searching but I know my qmk version is a bit older

Maartenwut, 2020-03-04:
It would be an easy fix - just exit the bootloader once the key is released, instead of when it is pressed.

fauxpark, 2020-03-04:
You can do it if you're fast enough, but yeah... not ideal. I guess [qmk/lufa](https://github.com/qmk/lufa) would be a better place to raise this, as that's where the actual QMK-DFU code lives.

drashna, 2020-03-04:
lol, yeah, I've ... run into this as well.  

Maybe this should be solved by documentation in the meanwhile?

Maartenwut, 2020-03-04:
@fauxpark you're probably right, though I don't see the issues tab there

Duckle29, 2020-03-06:
@Maartenwut Can you test the change in this PR: https://github.com/qmk/lufa/pull/9 :)

nstetter, 2020-03-29:
@Duckle29 I tested the [PR](https://github.com/qmk/lufa/pull/9) and exiting the bootloader with any custom key does not work for me:

```
#define QMK_ESC_OUTPUT B6 // COL0
#define QMK_ESC_INPUT D4 // ROW3
```
Pressing the according key or directly shorting the pins while in the bootloader does nothing.