qmk/qmk_firmware #8307: [Feature Request] A glossary for the documentation

HMarxen, 2020-03-04:
Very often I want to look something up but I can't remember under which topic to find it. Especially, for more advanced or obscure features finding them would be sometimes a lot quicker with a glossary.

drashna, 2020-03-04:
Well, skully has a start on reorganizing the docs to make it easier to navigate.
https://github.com/qmk/qmk_firmware/pull/8252

But that said, part of this is a limitation of the documentation generation that is used.  It would ideally be automated. Otherwise, it would be manually created, and that can be a lot of work. 

But yeah, it would be nice.

skullydazed, 2020-03-05:
We do have a glossary: https://docs.qmk.fm/#/reference_glossary

Once #8252 is merged it should be easier for you to find the topic you're interested in.

HMarxen, 2020-03-06:
Yeah but that is not very detailed. For example, there was an experimental feature where the backlighting changes with typing speed and I don’t remember where it was in the documentation. A full glossary would help here.
Maybe script which fetches all headlines, orders them alphabetically and then links to them all on one site.

noroadsleft, 2020-03-06:
> Yeah but that is not very detailed. For example, there was an experimental feature where the backlighting changes with typing speed

[Velocikey - QMK Firmware Docs](https://docs.qmk.fm/#/feature_velocikey)

> Maybe script which fetches all headlines, orders them alphabetically and then links to them all on one site.

Sounds more like an index.

skullydazed, 2020-03-06:
> Yeah but that is not very detailed.

The current glossary does have some missing terms, and we encourage people to click "edit this page" at the bottom to submit new entries when they notice an oversight.

> Maybe script which fetches all headlines, orders them alphabetically and then links to them all on one site.

Something like that sounds like a neat idea. If someone implements that we'd accept it.

skullydazed, 2020-03-06:
Also, we just merged #8252, please check out the current docs site (you may have to shift-reload for the updates to appear,) the sidebar has been reorganized so that features are easier to find.

HMarxen, 2020-03-07:
OK turns out I didn't know what a glossary is.^^
I meant an alphabetical index site, so when you know what you are looking for but not under which main category to find it, you can use the index. Unfortunately I am not a great programmer.

christopherjanzen, 2020-03-12:
@HMarxen wouldn't the search basically fulfill this need? If you know the term you can just type it in and get the filtered results. I'm wondering what a glossary would add that the TOC and search don't already provide?

closed 2020-07-10 by stale[bot]