qmk/qmk_firmware #8317: [Bug] Unnecessary layer activation delay when using "Layer Tap-Toggle": design bug?

Hexstream, 2020-03-05:
Hello,

I started using my Planck EZ 2 days ago, [here is my configuration](https://configure.ergodox-ez.com/planck-ez/layouts/ABWNG/latest/0).
Note the `TT 4` key at the bottom right on the base layer.

Within seconds of first trying this feature, I immediately knew that something is wrong with it, because when pressing the key there is an unexpected delay before the layer is activated. By artificially increasing the Tapping Term, I was able to confirm that the layer is indeed only activated once the Tapping Term has expired. This makes no sense, though, and the layer should immediately activate when pressing the TT key, regardless of the Tapping Term, just the same as for MT keys.

Here is my desired behavior for this feature, and I think it is entirely logical and without drawbacks, let me know if I missed anything. When I press a TT key, the layer should activate immediately and there are 3 possibilities:

1. If I do not press anything else and release the TT key before the Tapping Term expires, then the layer should remain activated until I press the TT key again.

2. If I do not press anything else and DO NOT release the TT key before the Tapping Term expires, then the layer should remain activated only until I release the TT key.

3. If I press anything else while still holding the TT key, then the layer should remain activated only until I release the TT key, regardless of the Tapping Term.

I believe this covers all cases. Let me know if I missed anything. I believe the current slight delay before activating the layer when pressing a TT key serves no useful purpose whatsoever and should thus be removed, thereby resulting in a much snappier user experience.

vomindoraan, 2020-03-30:
Case 1 is further regulated by the value of `TAPPING_TOGGLE`. Namely, the layer should remain activated only if the TT key has been tapped (pressed and released before the tapping term expires) `TAPPING_TOGGLE` times.

Possibility 3 is covered by adding `#define PERMISSIVE_HOLD` to your `config.h`. 

I agree that the initial layer activation could be moved to occur on the initial downstroke. However, the time of the layer's activation doesn't matter much before you actually try to press a key on that layer. Therefore, I think `PERMISSIVE_HOLD` already covers this use case successfully. The only noticeable downside would be if you were, say, using LEDs or underglow as a layer indicator.

Hexstream, 2020-03-30:
> The only noticeable downside would be if you were, say, using LEDs or underglow as a layer indicator; in that case you'd be able to notice that the layer wasn't being activated immediately upon pressing the TT key.

Bingo.

I provided a comprehensive outline of a full, perfect solution, so I'm not sure why we'd settle for anything less than a perfect solution. I'd like to provide a PR, but I suck at C and I'm working on much more important projects, so this is on the back burner for me. I'm confident this bug should be fairly easy to fix for someone knowledgeable in C and this codebase.

vomindoraan, 2020-03-30:
> I provided a comprehensive outline of a full, perfect solution, so I'm not sure why we'd settle for anything less than a perfect solution.

You're _almost_ correct; the _almost_ being that `TAPPING_TOGGLE` needs to be taken into account as I mentioned, which changes your described logic slightly. I will take a look at the code that handles TT and see what can be done to address this. I use underglow as a layer indicator on a few boards, so I would also like to see this behavior improved upon.

sigprof, 2020-05-03:
> Possibility 3 is covered by adding `#define PERMISSIVE_HOLD` to your `config.h`.

Note that `PERMISSIVE_HOLD` does not completely cover the possibility 3 — it covers only the following sequence:
* The TT key is pressed
* Another key (say, X) is pressed
* The X key is released
* The TT key is released

The important part is that the other key that is pressed during the tapping term must also be released before the TT key.

However, if those keys are released in a different order (and fast enough, so that the TT key is held less that the tapping term):
* The TT key is pressed
* Another key (say, X) is pressed
* The TT key is released
* The X key is released

then the TT action will be handled as a tap, not a hold; `PERMISSIVE_HOLD` does not handle this case.

In #8971 I proposed adding an option which would be even stronger than the current `PERMISSIVE_HOLD` (choose the hold action when any other key is just pressed, not pressed and then released); however, even adding such an option will not completely solve the delay issue (looks like the only way to process the initial press of tapping keys handled by the tapping code inherited from TMK is to do it inside `process_record_tap_hint()`, which currently handles only “swap hands” actions).

SuperCuber, 2021-04-23:
@vomindoraan I recently came across this behavior and would like it changed as well, have you looked at this since your reply?