qmk/qmk_firmware #8323: [Bug] Mouse key feature prevents Windows 10 from entering sleep

stbohne, 2020-03-06:
With MOUSEKEY_ENABLE = yes when I press the power button my laptop goes to sleep only to wake up a few seconds later.

## Describe the Bug

The Event Log shows for example:

Information	2020/03/06 19:57:18	Kernel-Power	507	(158)	System
The system is exiting connected standby 

Reason: Input Mouse.
Information	2020/03/06 19:57:13	Kernel-Power	506	(157)	System
The system is entering connected standby 

Reason: Power Button.

Setting MOUSEKEY_ENABLE = yes seems to fix the issue.

## System Information

 - Keyboard: Kyria Rev 1.1, Viterbi Rev 1.0
 - Operating system: Windows 10 Pro build 18363
 - AVR GCC version: avr-gcc.exe (GCC) 8.3.0

 - ARM GCC version: arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 - QMK Firmware version: 0.6.190

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

yiancar, 2020-03-06:
Sorry you are saying you have the issue with :
`MOUSEKEY_ENABLE = yes `
and that  `MOUSEKEY_ENABLE = yes` fixes your issue?:)

stbohne, 2020-03-07:
`MOUSEKEY_ENABLE = no` fixes it. But of course then I lose mouse keys.

tzarc, 2020-03-10:
Maybe it's something as simple as not sending a report if there's no change?

stbohne, 2020-03-27:
I've tracked it down to a call to `mousekey_send()` in `clear_keyboard_but_mods_and_keys()` in tmk_core/common/actions.c. Commenting that call out seems to avoid the issue without affecting mousekey functionality.

`clear_keyboard_but_mods_and_keys()` gets called in all sorts of different situations, so I have no idea what negative effects this has. For example, I noticed that no-op mouse events get sent when the layer is changed.

andrebrait, 2022-01-20:
Just chiming in, this is also the case on my Ubuntu laptop

arrowtype, 2024-01-30:
I’m also experiencing this on a RP2040-based keyboard, running vial-qmk. When I try to put my computer to sleep, while this keyboard is connected, the computer wakes instantly. I have to put the computer to sleep about 3 times for it to actually work.

> Setting `MOUSEKEY_ENABLE = no` seems to fix the issue.

Same here. This setting, when placed in my QMK keymap’s `rules.mk`, does workaround the issue. However, that means I can’t use the encoder on the keyboard for scrolling, which is a huge downside.

Is there any other potential solution here?

arrowtype, 2024-01-31:
Oh, actually, if I explicitly set `MOUSEKEY_ENABLE = yes`, then this issue goes away for me.

arrowtype, 2024-02-04:
Upon further testing, `MOUSEKEY_ENABLE` seems to have no impact on my Mac sleep function. The problem exists whether or not I use `MOUSEKEY_ENABLE`. I’m still trying to figure out a pattern, however.

drashna, 2024-10-13:
Is this still occurring?

andrebrait, 2024-10-14:
I haven't tried again. Will see if I can get the chance to do that the coming days.

arrowtype, 2024-10-15:
This still occurs for me on macOS 14.4, with either QMK or ZMK keyboards. My guess is, it’s a general issue of custom keyboards and macOS. I haven’t updated to macOS 15 yet, though, and it is possible it’s been fixed there.