qmk/qmk_firmware #8328: Compile breaks when you have empty layers in your JSON file [Bug]

alorda, 2020-03-06:
`.json` file from QMK Configurator does not compile correctly with `qmk compile` when there are empty layers before the final mapped layer.

## Describe the Bug
1. I created a `.json` on [QMK Configurator](https://config.qmk.fm) that had the following layer setup: 
  * 0 through 3 had keys configured
  * 4-9 were empty (they were empty arrays)
  * 10 had all my Fn keys in it
2. I exported it to my computer because the Online Configurator has been very slow today
3. I set up QMK Firmware on my Mac with:
  * Python 3.x (the latest as of today)
  * Ran `qmk setup`
  * Ran `qmk compile <name of my json file>`

Compile errors, and no creation of a `.hex` file.

```
ompiling: keyboards/idobo/idobo.c                                                                  [OK]
Compiling: keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.c            keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.c:14:26: error: macro "LAYOUT_ortho_5x15" requires 75 arguments, but only 1 given
  [4] = LAYOUT_ortho_5x15(),
                          ^
keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.c:14:8: error: 'LAYOUT_ortho_5x15' undeclared here (not in a function)
  [4] = LAYOUT_ortho_5x15(),
        ^~~~~~~~~~~~~~~~~
keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.c:15:26: error: macro "LAYOUT_ortho_5x15" requires 75 arguments, but only 1 given
  [5] = LAYOUT_ortho_5x15(),
                          ^
keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.c:16:26: error: macro "LAYOUT_ortho_5x15" requires 75 arguments, but only 1 given
  [6] = LAYOUT_ortho_5x15(),
                          ^
keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.c:17:26: error: macro "LAYOUT_ortho_5x15" requires 75 arguments, but only 1 given
  [7] = LAYOUT_ortho_5x15(),
                          ^
keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.c:18:26: error: macro "LAYOUT_ortho_5x15" requires 75 arguments, but only 1 given
  [8] = LAYOUT_ortho_5x15(),
                          ^
keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.c:19:26: error: macro "LAYOUT_ortho_5x15" requires 75 arguments, but only 1 given
  [9] = LAYOUT_ortho_5x15(),
                          ^
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [.build/obj_idobo_jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keyboards/idobo/keymaps/jdev-id75-bigbang-qw-dvk-fullbalancedmargin/keymap.o] Error 1
make: *** [idobo:jdev-id75-bigbang-qw-dvk-fullbalancedmargin] Error 1
Make finished with errors
```
## System Information

 - Keyboard:  Idobo 75
   - Revision (if applicable):  
 - Operating system: Mac OS
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Generating `.hex` files from the online QMK Configurator with empty layers was never an issue.

devilzmods, 2020-07-06:
to fix this manually, go into your keymap.c that is created by the .json compiler (it tells you the location at the beginning) and just delete the empty layer's code lines.

I'm not a coder, but qmk configurator should probably fill empty layers with KC_NO, or even better the json to .c converter should skip empty layers instead of not providing arguments.

[image]
broken keymap.c
[image]
fixed it.

fauxpark, 2020-07-06:
I think a more solid solution would be for the Configurator to disallow nonconsecutive layers in the first place, as doing this pointlessly takes up flash space. So this would be an issue for https://github.com/qmk/qmk_configurator.

Nadrieril, 2020-07-06:
I wouldn't want that: I have some upper layers that I rarely change and some lower layers that I often add or remove.  I enjoy not needing to shift those upper layers as well as any code that mentions them whenever I add or remove a lower one.

devjona, 2020-10-19:
> 
> 
> I wouldn't want that: I have some upper layers that I rarely change and some lower layers that I often add or remove. I enjoy not needing to shift those upper layers as well as any code that mentions them whenever I add or remove a lower one.

I'm with you on this; there is a good use-case for leaving some empty layers to allow for future addition of layers, while leaving the top layer in the same index and not having to remap all the keys that access this top layer.