qmk/qmk_firmware #8360: [Bug] There's no way to tell Space Cadet to not emit the tapping key

skorokithakis, 2020-03-09:
## Describe the Bug
 
I have commented in #3192 but I think this deserves its own issue.

It's currently a problem that it's impossible to tell Space Cadet that you don't need the tapping key and to not emit it. You can `return false` in the processing function for other keys, but there's no way to cancel Space Cadet's tapping.

My usage is:

* Space Cadet shifts (pressing is parenthesis, holding is shift).
* Alt+Shift passing through as normal, for language change in the OS (no parentheses, no matter whether I instantly tap Shift).
* Left shift+Right shift sending "()", as sometimes I haven't managed to release left shift before tapping right.
* Right shift+Left shift sending Caps Lock.

I currently can't make these work, because if I tap a Shift key quickly, its tapping keycode is always emitted, even if it was part of Alt+Shift for changing the language, and there's no way to change this from my config because I can't tell Space Cadet to not emit the parenthesis no matter how short the tap is.

There should be a way to tell Space Cadet "you don't need to emit the tapping key, I've got it".

## System Information

 - Keyboard:  Centromere
   - Revision (if applicable): v2 
 - Operating system: Linux
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version: I'm using @spe2's fork, I'm not entirely sure what version it is, but I would love to use mainline.

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

Erovia, 2020-03-16:
Have you tried tuning `TAPPING_TERM`?

skorokithakis, 2020-03-16:
Yes, it's set to 200. That's not the issue, the problem is that you can't tell QMK to not emit the tap.

jemag, 2021-07-23:
Just to clarify, when you press Left shift + Right shift and don't fully release Left shift, is it sending both parentheses '()'  or only the right one at the moment?

Right now, for me, I cannot send both parentheses when left shift isn't fully released which makes space cadet shift near useless for programming and calling functions without parameters while typing fast (i.e: someFunc())

skorokithakis, 2021-07-23:
It's been a year, I don't remember what it used to do before. I ended up writing a custom function to send `()` when I pressed left-then-right.

jemag, 2021-07-23:
could you share your custom function? This is exactly what I am trying to do, please!

skorokithakis, 2021-07-23:
@jemag you are not a lucky man, I have the config on my other PC. I will be able to send it on Monday, if I don't forget.

skorokithakis, 2021-07-30:
This is my code:

```
static bool shift_interrupted[2] = {0, 0};
static uint16_t scs_timer[2] = {0, 0};
static bool caps = false;

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
    case KC_RSPM: {
        if (record->event.pressed) {
            if (get_mods() & MOD_BIT(KC_LSHIFT)) {
                shift_interrupted[0] = true;
                shift_interrupted[1] = true;
                SEND_STRING("()");
            } else {
                shift_interrupted[1] = false;
                scs_timer[1] = timer_read();
                register_mods(MOD_BIT(KC_RSFT));
            }
        } else {
            if (get_mods() & MOD_BIT(KC_LSFT)) {
                shift_interrupted[0] = true;
                shift_interrupted[1] = true;
            }
            if (!shift_interrupted[1] && timer_elapsed(scs_timer[1]) < TAPPING_TERM) {
                register_code(RSPC_KEY);
                unregister_code(RSPC_KEY);
            }
            unregister_mods(MOD_BIT(KC_RSFT));
        }
        return false;
    }
}
```