qmk/qmk_firmware #8375: Can atmega32a drivers ws2812 by bitbang?

tinggua, 2020-03-11:
Hello, I have made a keybord with atmega32a. It has the rgb underglow, and i connected ws2812s with mcu's io pin directly(just like many atmega32u4).But now i find the ws2812s can not work. So is 32a not supported bitbang driver the ws2812?
Before that  i find ps2avrgb bords used atmega32a, and use i2c to driver the ws2812 ,but i can not find circuit diagram of them for eference ,where can i find that?
Thank you for your answer .

mtei, 2020-03-11:
I feel it is clock speed matter.

https://github.com/qmk/qmk_firmware/blob/2b66acf04a82bc571ac5f545be7c43af722d6704/drivers/avr/ws2812.c#L93-L101

yulei, 2020-03-11:
ps2avr used an attiny to drive ws2812, so you need to buy this from author of ps2avr or write your own driver on attiny to translate i2c commands from atmega32a to ws2812 signal sequence. If your made the keyboard by yourself, why not just using atmega32u4?

yiancar, 2020-03-28:
yes they can, they work just fine and I have tried this.
I think this issue can be closed

zvecr, 2020-04-01:
From #8598, I had to modify the existing AVR implementation. 32a partially shares the same register config so i cant see how the existing code would work without the same changes.

yiancar, 2020-04-01:
is this about entering the bootloader?

zvecr, 2020-04-26:
Verified working on <https://www.rgbkb.net/products/pan-keyboard-diy-kit> by Legonut.

closed 2020-04-26 by zvecr