qmk/qmk_firmware #8376: Allow for more index of than 128 for XP() Unicode entry

SpaceCadetSkywalker, 2020-03-11:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

In documentation (https://docs.qmk.fm/#/feature_unicode?id=lower-and-upper-case) it is described that `XP(i, j)` can only operate on the first 128 items defined in `unicode_map`.

> Due to keycode size constraints, i and j can each only refer to one of the first `128` characters in your unicode_map. In other words, `0 ≤ i ≤ 127` and `0 ≤ j ≤ 127`. This is enough for most use cases, but if you’d like to customize the index calculation, you can override the `unicodemap_index()` function. 

How to modify this function to operate on (say) `256` or `512` of the initial items in the `unicode_map` thus allowing for more Unicode symbols to be available with shift modifier?

Alternative formulation or simple example in the documentation would be much appreciated by folks who intend to use multiple symbols etc.

fauxpark, 2020-03-13:
This is not really possible at the moment, unfortunately. As the documentation states, it's because the size of a QMK keycode does not allow for it.

Let's look at how the `XP()` keycode is defined in quantum_keycodes.h. First off, a keycode in QMK is an unsigned 16-bit integer.

The range of `XP()` is given by the `quantum_keycodes` enum at the top of the file:
```c
    QK_UNICODEMAP_PAIR     = 0xC000,
    QK_UNICODEMAP_PAIR_MAX = 0xFFFF,
```
and the keycode itself further down:
```c
#    define XP(i, j) (QK_UNICODEMAP_PAIR | ((i)&0x7F) | (((j)&0x7F) << 7))  // 127 max i and j
```

`0xC000` is `1100 0000 0000 0000` in binary - those two top bits have to stay 1s to identify the keycode as an `XP()`, so we now only have 14 bits of usable range.
Since it takes two parameters, `i` and `j`, which are indexes into the `unicode_map` array, we split those 14 bits equally. Thus each parameter has 7 bits available to them, and to calculate the range of numbers you can get with x number of bits, we do 2^7 = 128.
In effect, the keycode uses this format in binary:
```
11ii iiii ijjj jjjj
```

I hope this is clearer :)

closed 2020-07-11 by stale[bot]

ChMunk, 2022-02-20:
Instead of storing both parameters independently in the keycode, wouldn't it make sense only to store `i` as the full index, and then let `j` be represented as its difference from `i`? That way, you could restrict the difference quite easily, making sure `i` and `j` are close to each other on the list. Say, 4 bits of wiggle room (16 options, perhaps from -8 to +8). That way, the binary keycode could look like this:

`11ii iiii iiii dddd`

Then there would be 10 bits for the index of `i` (the first 1024 items on the list). Or even more, if you restrict the difference more.