qmk/qmk_firmware #8381: [Bug] implicit declaration of function

C0DK, 2020-03-11:
I'm trying to implement layer-base underglow.

When i import `#define RGBLIGHT_LAYERS` i get a bunch of compile errors.
## Describe the Bug
I am following https://docs.qmk.fm/#/feature_rgblight
I added the following to `config.h`:
```
#define RGBLIGHT_LAYERS
```
and naturally the following to `rules.mk`:
```
RGBLIGHT_ENABLE = yes
```
but upon make'ing, iget the following errors - which seems to be errors in the framework itself (and not due to my implementation, as it doesn't bug out without the `#define RGBLIGHT_LAYERS`):
```
Compiling: quantum/rgblight.c                                                                      quantum/rgblight.c: In function 'rgblight_layers_write':
quantum/rgblight.c:645:49: error: implicit declaration of function 'pgm_read_ptr'; did you mean 'pgm_read_word'? [-Werror=implicit-function-declaration]
  645 |         const rgblight_segment_t *segment_ptr = pgm_read_ptr(layer_ptr);
      |                                                 ^~~~~~~~~~~~
      |                                                 pgm_read_word
quantum/rgblight.c:645:49: error: initialization of 'const rgblight_segment_t *' {aka 'const struct <anonymous> *'} from 'int' makes pointer from integer without a cast [-Werror=int-conversion]
quantum/rgblight.c:652:13: error: implicit declaration of function 'memcpy_P'; did you mean 'memcpy'? [-Werror=implicit-function-declaration]
  652 |             memcpy_P(&segment, segment_ptr, sizeof(rgblight_segment_t));
      |             ^~~~~~~~
      |             memcpy
cc1: all warnings being treated as errors
 [ERRORS]
```

I tried different things and among others `make mxss:default`, which gave me header errors as well:
```
Compiling: keyboards/mxss/rgblight.c                                                               In file included from keyboards/mxss/rgblight.c:30:
keyboards/mxss/rgblight.h:19: error: "pgm_read_word" redefined [-Werror]
   19 | #define pgm_read_word(addr) (*(const unsigned short *)(addr))
      |
In file included from tmk_core/common/progmem.h:4,
                 from keyboards/mxss/rgblight.c:28:
/usr/avr/include/avr/pgmspace.h:991: note: this is the location of the previous definition
  991 | #define pgm_read_word(address_short)    pgm_read_word_near(address_short)
      |
In file included from keyboards/mxss/rgblight.c:30:
keyboards/mxss/rgblight.h:20:10: fatal error: rgblight_reconfig.h: No such file or directory
   20 | #include "rgblight_reconfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
compilation terminated.
 [ERRORS]
```
the reconfig file seems, though, to be the file that would define the missing `pgm_read_ptr` function. So these bugs seem related. 

I wanted to dig into the source and provide an actual bugfix but i'm not that great at C and don't know your codebase well, so i would be pretty useless. Also it might just be my config or something that is bad.

For reference, my keymap can be found here:

https://github.com/C0DK/dotfiles/tree/master/keymaps/cabang

## System Information

 - Keyboard:  
   - Revision (if applicable):  Preonic Rev3
 - Operating system: 4.19.107-1-MANJARO
 - AVR GCC version: `9.2.0`
 - ARM GCC version: `9.2.0`
 - QMK Firmware version: `0.8.26`

zvecr, 2020-03-11:
mxss rgblight header issue should be fixed with #8388 (this wont fix the rgblight layers issue).

C0DK, 2020-03-12:
@zvecr So the issues should be independent? fair enough.

C0DK, 2020-03-13:
I think i found a solution. It's not dry but it does solve the problem. Maybe someone more experienced can correct where i should place my solutions if i make a PR. will do that soon ish

PR #8406 (2020-03-28): fixed problem with implicit declaration in quantum/rgblight.c
## Description

So i am well aware that my solution probably isn't up to standards - neither code nor modularity nor DRY principles - i don't normally work with C, but had a bug that made the layers stuff useless. I fixed this for my working version and would just, as a friendly gesture, provide you guys with the solution. I hope you can possibly guide me to a better way of fixing the bug (maybe by importing the correct header, or something else)

I also want to say that this introduces a warning, which i had to ignore, but it works [... cut]
closed 2020-03-28 by zvecr