qmk/qmk_firmware #8384: Can't find PM port

adcampbell, 2020-03-11:
I don't know if this is better off posted in a pro micro forum or not, so my apologies if this is the wrong place.

I have a pro micro on a redox. I initially used the cli to flash the default redox to the board, and got a success message. None of the keys worked, so I edited the hex file and tried again. This time, the COM port would not show up when I reset the board. Instead, the board shows up as a "USB Composite Device". No program (cli, qmk toolbox, avrdudess) can find it to flash it. I tried uninstalling the drivers, redoing it on linux, and various other things, and cannot get the PM to be reached. Is there a way to clear it and start over? It's an atmega32u4 if that wasn't clear. I tried searching and couldn't find this particular issue anywhere. Thank you for your time.

Erovia, 2020-03-12:
Hi!
What do you mean by 

> edited the hex file and tried again

Could you please describe what and how you did this?

adcampbell, 2020-03-13:
Sorry, I meant that I edited the keymap and created a new hex file - so not the default redox one.

Erovia, 2020-03-16:
What happens if you reset it? How is it identified in the `lsusb` output?
It should be something like `2341:0036 Arduino SA Arduino Leonardo`

adcampbell, 2020-03-18:
I'm using virtualbox to run linux in a vm. I have to attach a usb item. When I attach it and run lsusb the output is "Bus 001 Device 005: ID feed:0000". If I reset it virtualbox drops the attachment. 

I'm using a GUI application called USBDeview by nirsoft to see it using windows. Please see attached picture. "The Redox Keyboard" at the bottom seems to be this microcontroller. 
[image]

fauxpark, 2020-03-18:
That device is the actual keyboard; what you want is the bootloader device which will enumerate with a different VID and PID when you short RST and GND. Pro Micro bootloaders will only stay in bootloader mode for only about 8 seconds so you'll want to set up the VirtualBox passthrough beforehand.

adcampbell, 2020-03-19:
I set it up but it just disappears from lsusb output when i reset it.

fauxpark, 2020-03-19:
What does your passthrough device look like? Note that when you reset, it will show up as a completely different device, so if you've set it up to look for the FEED:0000 device, that won't work. For example:
[image]

adcampbell, 2020-03-19:
[image]

fauxpark, 2020-03-19:
It should turn up in Device Manager, under Ports.
[image]

Erovia, 2020-03-19:
The problem is that once you reset the PM, it will show up as a new device on the host which is not yet passed through to the vm. But the PM only stays in bootloader mode for a few secs, so by the time you pass it through, it already booted into firmware.

Imho, you'll have an easier time if you flash it on either Windows or boot into a Linux liveusb.

adcampbell, 2020-03-19:
@Erovia I flashed it on windows the first time, running qmk on linux was just my attempt to fix my issue. Though now I know it won't work easily on my linux vm.

@fauxpark The crux of my issue is that it will never show up under ports when resetting. No program or CLI has been able to find it.

adcampbell, 2020-03-25:
At this point I just believe it to be a faulty PM. I gave up and ordered a few new ones and had zero issues with them when following the same steps. Closing

closed 2020-03-25 by adcampbell