qmk/qmk_firmware #8391: TADA68 QMK-DFU bootloader jump does not work [Bug]

tdgrunenwald, 2020-03-12:
## Description
After reading #7679 and #1215 I was hopeful to get a setup similar to what @tokyovigilante had achieved. When I flash tada68_tokyovigilante_production.hex, the keymap works, except it becomes unresponsive after RESET key is hit, and does not appear as DFU device. The bootloader exit key also does not return it to the user program. Re-connecting the usb cable returns the keyboard to normal/user program.

## System Information

 - Keyboard: TADA68
   - Revision (if applicable): Unknown
 - Operating system: Void GNU/Linux, glibc variant w/ kernel 5.4.20_1
 - AVR GCC version: 9.2.0

 - QMK Firmware version: 0.7.103, 0.8.20

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
Initially, when I try to make the keymap from either the latest on master or exact commit where #7679 was merged (~0.7.103), I use the target `make tada68:tokyovigilante:production` and it fails with the following avr-ld error:
```
 [LNK]     : Linking object files into "BootloaderDFU.elf"
avr-gcc obj/BootloaderDFU.o obj/Descriptors.o obj/BootloaderAPI.o obj/BootloaderAPITable.o obj/HIDParser.o obj/Device_AVR8.o obj/EndpointStream_AVR8.o obj/Endpoint_AVR8.o obj/Host_AVR8.o obj/PipeStream_AVR8.o obj/Pipe_AVR8.o obj/USBController_AVR8.o obj/USBInterrupt_AVR8.o obj/ConfigDescriptors.o obj/DeviceStandardReq.o obj/Events.o obj/HostStandardReq.o obj/USBTask.o -o BootloaderDFU.elf -lm -Wl,-Map=BootloaderDFU.map,--cref -Wl,--gc-sections -Wl,--relax -mmcu=atmega32u4 -Wl,--section-start=.text=0x7000 -Wl,--section-start=.apitable_trampolines=0x7FA0 -Wl,--undefined=BootloaderAPI_Trampolines -Wl,--section-start=.apitable_jumptable=0x7FE0 -Wl,--undefined=BootloaderAPI_JumpTable -Wl,--section-start=.apitable_signatures=0x7FF8 -Wl,--undefined=BootloaderAPI_Signatures
/usr/bin/avr-ld: section .apitable_trampolines LMA [0000000000007fa0,0000000000007fb7] overlaps
section .data LMA [0000000000007f5e,0000000000007fc7]
collect2: error: ld returned 1 exit status
make[2]: *** [../../LUFA/Build/DMBS/DMBS/gcc.mk:250: BootloaderDFU.elf] Error 1
make[2]: Leaving directory '/home/tyler/firmwares/qmk/lib/lufa/Bootloaders/DFU'
make[1]: *** [tmk_core/avr.mk:328: bootloader] Error 2
make[1]: Leaving directory '/home/tyler/firmwares/qmk'
Make finished with errors
make: *** [Makefile:579: tada68:tokyovigilante:production] Error 1
```
I am able to get it to compile successfully by setting `BOOTLOADER_SIZE = 6144` in the bootloader.mk file for the qmk-dfu bootloader (setting this in the keymap's rules.mk file appears to have no affect).

I flash with the command `avrdude -F -c pi_1 -p m32u4 -e -U flash:w:tada68_tokyovigilante_production.hex:a -U lfuse:w:0x5E:m -U hfuse:w:0xD9:m -U efuse:w:0xC3:m -U lock:w:0x3F:m`.
This reports a validation error after writing the lock `0xff != 0x3f`, but I read that this is because the most significant bits are unused and default to 1, so that adds up I think.

I think it is something to do with where the bootloader is written in memory and where the RESET is attempting to jump, but I lack experience with qmk and microcontrollers in general.

tokyovigilante, 2020-03-12:
GCC 9.2 is not compatible with QMK as per #6719. I use 7.x, but I understand 8.3 works well too.

tdgrunenwald, 2020-03-12:
Ahh, I should have checked that. Thanks for the quick help. I am building gcc 8.3 now, should be able to test it again this afternoon

tdgrunenwald, 2020-03-13:
avr-gcc 5.4.0 worked for me. Turns out, building gcc from sources is not as easy as it looks...Anyway, keyboard works perfectly now. Thanks!

closed 2020-03-13 by tdgrunenwald