qmk/qmk_firmware #8420: [Feature Request] Support HT32F1755 ARM Cortex-M3 (CM Storm QFR-i)

B3QL, 2020-03-14:
It would be great to have support for [CM Quick Fire RAPID-i ](https://www.coolermaster.com/catalog/peripheral/keyboards/quick-fire-rapid-i/).

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior
- [x] New hardware support

## Description
Microcontroller: Holtek HT32F1755 [Datasheet](https://www.holtek.com.tw/documents/10179/11842/HT32F1755-1765-2755_Datasheetv120.pdf)
Custom update protocol: http://spritesmods.com/?art=rapidisnake&page=4

tzarc, 2020-03-23:
This is unlikely, unless someone is able to add support for the MCU and get it merged into the upstream https://github.com/ChibiOS/ChibiOS-Contrib repository. QMK does not have the resources to maintain their own set of MCU definitions.

closed 2020-07-21 by stale[bot]

ppkt, 2020-08-20:
@B3QL have you seen these repos?
https://github.com/pok3r-custom/pok3r_re_firmware
https://github.com/pok3r-custom/qmk_pok3r

pok3r keyboards are using the same MCU as CM Storm - HT32 - and they even patched ChibiOS to add support for this MCU:
https://github.com/pok3r-custom/ChibiOS-Contrib

B3QL, 2020-08-25:
@ppkt No I haven't seen those projects. I'll definitely check them out! Thanks ;)