qmk/qmk_firmware #8535: [Bug] set_single_persistent_default_layer doesn't persist after Drop CTRL loses power

nicksnyder, 2020-03-23:
## Describe the Bug

I have two keymaps that I want to switch between (one has a Mac layout for cmd/alt and the other has a windows layout for cmd/alt). Ideally I want the chosen default keymap to persist when the keyboard loses power.

The documentation on [switching and toggling layers](https://docs.qmk.fm/#/feature_advanced_keycodes?id=switching-and-toggling-layers) says:

> DF(layer) - switches the default layer. The default layer is the always-active base layer that other layers stack on top of. See below for more about the default layer. This might be used to switch from QWERTY to Dvorak layout. (Note that this is a temporary switch that only persists until the keyboard loses power. **To modify the default layer in a persistent way requires deeper customization, such as calling the set_single_persistent_default_layer function inside of process_record_user.**)

I added keycodes to activate my keymaps and called `set_single_persistent_default_layer` inside of `process_record_user` as shown below:

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    static uint32_t key_timer;

    switch (keycode) {
        case WIN_KEY:
            if (record->event.pressed) {
                set_single_persistent_default_layer(WINDOWS_KEYMAP);
            }
            return true;
        case MAC_KEY:
            if (record->event.pressed) {
                set_single_persistent_default_layer(MAC_KEYMAP);
            }
            return true;
// code below here not included in this snippet
```

This successfully switches the keymaps, but when I disconnect power, it reverts to the default keymap at index 0 (the Windows keymap in my case). Given the documentation, I was under the impression that `set_single_persistent_default_layer` would cause the selected default layer to persist even if the keyboard loses power. Is the documentation incorrect?

## System Information

 - Keyboard:  Drop CTRL
   - Revision (if applicable):  
 - Operating system: macOS Catalina 10.15.3
 - AVR GCC version: 8.3.0
 - ARM GCC version: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 - QMK Firmware version: 0.8.58

 - Any keyboard related software installed? No

tzarc, 2020-03-23:
The Massdrop boards have no non-volatile storage at this point, so this is a known issue.
We're waiting for Massdrop/mdloader#16 to go through before we can consider adding support for it.

nicksnyder, 2020-03-23:
Got it, thanks for the links!

zvecr, 2021-12-17:
Fixed with 6068.

closed 2021-12-17 by zvecr