qmk/qmk_firmware #8541: [Bug] qmk setup fails with `ValueError`

kdheepak, 2020-03-23:
The output of avr-gcc is 

```
avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

in https://github.com/qmk/qmk_firmware/blob/b936048b0bf2b5508104e243089e785debc431d0/lib/python/qmk/cli/doctor.py#L51 using 

        version_number = first_line.split()[-1]

instead works for me.

StreakInTheSky, 2020-03-24:
Found this bug as well, and did the same fix. Not sure if it's strictly a mac/homebrew issue. Which could explain why this was missed.

fauxpark, 2020-03-24:
This might be because the version string changed, so the item at index 2 is now the string "AVR" instead of the version. Perhaps we should use the `-dumpversion` flag instead:

```
fauxpark@marissa:~$ avr-gcc -dumpversion
8.4.0
fauxpark@marissa:~$ arm-none-eabi-gcc -dumpversion
8.3.1
```

StreakInTheSky, 2020-03-28:
Tried changing `is_executable` to run with -dumpversion instead, but the other items in `ESSENTIAL_BINARIES` return status codes greater than 1. So would have to either change the only flag for those two items, or just change how it checks the string for avr-gcc, which would be easier.

commit 3fad3854 (2020-04-04): Fixes #8541 by getting version from -dumpversion then --version as fallback
closed 2020-04-04 by skullydazed