qmk/qmk_firmware #8542: [Bug] Linker error

kdheepak, 2020-03-23:
I'm getting a linker error when trying to compile the firmware for the keyboard.

```
$ qmk compile -kb helix -km kdheepak

Ψ Compiling keymap with make helix:kdheepak

QMK Firmware 0.8.71
Making helix/rev2 with keymap kdheepak

avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Linking: .build/helix_rev2_kdheepak.elf                                                             [ERRORS]
 |
 | /Users/USER/.local/opt/avr-binutils/bin/avr-ld: cannot find crtatmega32u4.o: No such file or directory
 | /Users/USER/.local/opt/avr-binutils/bin/avr-ld: cannot find -latmega32u4
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/helix_rev2_kdheepak.elf] Error 1
make: *** [helix:kdheepak] Error 1
Make finished with errors
```

Any thoughts on what might be going on here?

## System Information

 - Operating system: MacOS 10.15.3
 - AVR GCC version: 
```
avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

 - ARM GCC version: 
```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
 - QMK Firmware version:

0.8.71

fauxpark, 2020-03-23:
This should be fixed by https://github.com/osx-cross/homebrew-avr/pull/183

ladislas, 2020-03-23:
Bottles with a fix are currently being built. I'll merge and publish tomorrow.

In the mean time, you can look up https://github.com/osx-cross/homebrew-avr/pull/185 for a fix if you want to build from source.

kdheepak, 2020-03-24:
I guess it might be related to this: https://github.com/osx-cross/homebrew-avr/issues/182

kdheepak, 2020-03-24:
I built from source after making the same changes from the patch and this worked!

closed 2020-03-24 by kdheepak