qmk/qmk_firmware #8567: Is it possible to have key-chords?

blasco, 2020-03-26:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Hi, I'm considering buying the Ergodox. I see a lot of interesting functionality such as:

Dedicated Hyper and Meh keys, Space-cadet Shift, Mouse Control, Dual-function Keys

I'm specially in love with the "Dual-function Keys", it was something that I never thought about before. 

I've been using vim for over 8 ages, and one thing that completely boosted my workflow was adding **key-chords** (also known as vim-arpeggios). 

The idea here is that you can simultaneously press two keys two obtain a different effect (it's quite aligned with the "Dual-functions Keys" functionality. This allows to create very powerful key combinations (it's like extending ctrl and shift to every single key on the keyboard) with the advantage of rich mnemonics:

Example:
In vim, if you press:
`s` := **s**ubstitute text 
`p` := **p**aste text
With my configuration, I can add a chord, so if I press both of them simultaneously: 
`s`+`p` := **s**earch in **p**roject (very ergonomic and easy to remember)

I wonder if the keyboard could have that functionality, so I could extend this to any program. We could map easy to remember key combinations to more complex ones:
`s`+`f` := **s**earch in **f*ile -> mapped to `ctrl-f`

Once you start using key-chords you cannot stop using them! I use lots of them and they literally allow to do two keystrokes at the time of just one.

Could this be achieved? I would love contributing to make this happen

blindsteal, 2020-03-27:
Hey there, I'm just getting started with QMK myself but did you check out [combos](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_combo.md)?
With this you should theoretically be able to map whatever Chord you want to whatever Keycodes you want.

blasco, 2020-03-27:
That's fantastic! It's exactly what I was looking for. Thank you very much :)

blasco, 2020-03-27:
Do you know it the combos can be added via the configuration tool? @blindsteal

zvecr, 2020-03-27:
No, its going to be a compile from source thing.

blindsteal, 2020-03-27:
Yeah @blasco sorry I didn't mention that, I've never used the configurator because I'm currently building from a fork anyway 😬
It's not too hard though, just follow the steps in the [tutorial section](https://docs.qmk.fm/#/newbs) 🙂

closed 2020-03-29 by blasco