qmk/qmk_firmware #8573: [Bug] BL_ controls on XD60 are inverted

E66666666, 2020-03-27:
On my XD60 rev2, the backlight controls are inverted. I'm using the QMK Configurator and flash the .hex file with QMK Toolbox. 

## Describe the Bug

Specifically:

* `BL_TOGG` toggles between full-brightness and user brightness
* `BL_INC` decreases brightness
* `BL_DEC` increases brightness

When the system is going to sleep/off, the keyboard brightness jumps to full brightness.

I have taken a quick look at the `config.h` and `rules.mk`, but didn't find anything aside from disabling backlight altogether.

## System Information

 - Keyboard:  XD60
   - Revision (if applicable): rev2
 - Operating system: Windows 10
 - AVR GCC version: 5.4.0 (QMK Configurator)

 - ARM GCC version: ?

 - QMK Firmware version: b9d0b1f06478a1b3f596882b33cb9bd84e477bea (current master)

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I'm not sure when it started to happen, as it's been almost a year since my last layout change.

fauxpark, 2020-03-27:
Could you try adding `#define BACKLIGHT_ON_STATE 0` to the config.h?

For context, this define controls the state the backlight pin should be in when the LEDs are considered to be "on". This has to do with the type of transistor being used in the backlight circuit.
Prior to our [last round of breaking changes](https://docs.qmk.fm/#/ChangeLog/20200229?id=adding-support-for-backlight_on_state-for-hardware-pwm-backlight) the define defaulted to 0 and was only respected for software PWM. Hardware PWM on the other hand (which is supported on only a few pins) was hardcoded to assume 1 = on. Since the vast majority of boards in QMK used hardware PWM capable pins for the backlight, the default was changed to minimise the impact.

It looks like this board has its backlight pin on F5, which is *not* a hardware PWM pin, so now the logic has been inverted.

E66666666, 2020-03-27:
Adding that line fixes the issue completely. Thanks for the context too, helps with understanding why it happens.

PR #8575 (2020-03-27): Fix inverted backlight for XD60

## Description

The XD60's backlight is on a non-hardware PWM pin, so the recent breaking change to the default value of `BACKLIGHT_ON_STATE`... has broken it.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #8573 

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the d [... cut]
closed 2020-03-27 by zvecr