qmk/qmk_firmware #8592: Allow split keyboards on chips without USB like 328p

triron-pixel, 2020-03-28:
Trying to make a split keyboard with a 328p chip. And the software does not compile due to the fact that split_util.c use USB registers. So make this patch for undef USB stuff.

[0001-split-for-chip-without-usb.patch.txt](https://github.com/qmk/qmk_firmware/files/4397818/0001-split-for-chip-without-usb.patch.txt)

chubylive, 2020-03-28:
I am trying to do something similar. I think there is more usb stuff that needs to either taken out or defined out.

zvecr, 2020-03-28:
@chubylive lufa isnt used on a 328p (or 32a) so thats the wrong rabbit hole.

I've raised #8594 but it still has transport requirements.
* serial is currently 32u4 only
* i2c compiles but is untested

yiancar, 2020-03-28:
i2c should work I use it for other things

zvecr, 2020-03-28:
@yiancar In this context, its i2c_slave that would potentially cause issues, and to my knowledge that is currently untested.

yiancar, 2020-03-28:
I have tested i2c_slave as well. mostly the AVR library takes cares of the abstraction here for most 8bit controllers

zvecr, 2020-03-28:
Good to know!. But given the solution as a whole doesn't compile, i'm still going to stand by my "i2c transport code is untested" stance.

yiancar, 2020-03-28:
yep I have no clue about the rest just that you can send with our libraries something from one 328p master to a slave

PR #8594 (2020-05-21): Use LUFA funcs for split_util

## Description

While in the area, i have also cleaned up the logic so that non LUFA avr boards can just implement `is_keyboard_master`.

(cb4a64e6c5013091c8864366320874f9114625c5 was added just to check compile issues, it can be removed if required)

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Partially fixes #8592

## Checklist

- [x] My code follows the code style o [... cut]
closed 2020-09-18 by stale[bot]