qmk/qmk_firmware #8593: [Bug] Console Descriptor shoud be one way

yiancar, 2020-03-28:
## Describe the Bug

Looking at the console descriptor it has both an input and output section.
This is probably because of the rawhid code from PJRC. In my understanding we use console as a one way communication tool. From the keyboard TO the host. If this is true, and if qmk-toolbox is only listening, I suggest we modify the descriptors to show that only output exists.

In a quick look it looks like console code does not handle any input anyway, please correct me if I am wrong

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

yiancar, 2020-03-29:
Ok initial check:
https://github.com/yiancar/qmk_firmware/commit/4ecea11aaf7e4457cabef0f9a806e364caa4a64c

Chibios:
Removed the output hid descriptor, removed the input handling. All good.
Trying to remove the endpoint is not as straight forward.

Removing `CONSOLE_IN_EPNUM` starts to cause issue as in usb_main.c the macro QMK_USB_DRIVER_CONFIG need the epnum. Furthermore `out_ep_config` is always in the macro. Any ideas how to exclude it without making everything ugly?

drashna, 2024-10-13:
This was fixed in 22304.

PR #22304 (2024-01-09): Remove console out endpoint

## Description

ChibiOS changes a bit gross as QMK_USB_DRIVER_CONFIG and the various driver code assumes both endpoints and unpicking it is much more of a change. Instead the vusb implementation was cribbed.

Tested on:
* chibios - drop/alt/v2 + zvecr/zv48/f401
* lufa - drop/thekey/v2
* vusb - jj4x4
* arm_atsam - compiles but ignored due to pending removal.

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issue [... cut]
closed 2024-10-13 by drashna