qmk/qmk_firmware #8595: Expose OLED buffer to userspace functions

brickbots, 2020-03-29:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
With recent changes to the OLED driver to enable raw buffer writing, I'd like to be able to read the current OLED buffer from user space code.  This will enable more sophisticated visualization routines in user space and I am seeking advice on the safest, most QMK-like way to do this.

I can think of at least two ways to enable read access to the OLED buffer from keymaps or user space code:

* Declare the buffer in oled_driver.h as extern, enabling a redefinition in other code files
* Provide a function in oled_driver.c to provide either a copy of, or pointer to, the buffer

I've played with both and settled on the extern solution in my own test branch, but before issuing a pull request for this change, I wanted to check with some smarter folks on the ramifications of either approach.

Thank you for any advice on the best way to implement this.

mtei, 2020-04-25:
Tagging @XScorpion2

XScorpion2, 2020-04-25:
I think extern is the wrong approach to go about this as then the name of the buffer is now fixed, and cannot be changed if necessary due to collision. 

Initial thought (no coffee yet): maybe a function that takes some optional parameters, such as position you want to retrieve, then returns a pointer to that buffer location & size remaining in the buffer.

mtei, 2020-04-25:
@XScorpion2, See also #8777.

XScorpion2, 2020-04-25:
@mtei can I get coffee first? =D

closed 2020-08-23 by stale[bot]

brickbots, 2020-08-24:
Just a note for anyone following along at home, a merge request to address this #8777 has been merged into the development branch and should be universally available when this branch becomes the master.

Thanks everyone for your help on this one!