qmk/qmk_firmware #8600: Recent commit breaks LT(5,KC_J) on FC660C

Syzygies, 2020-03-29:
## Describe the Bug

I had fallen 27 commits behind master while working on several keyboards. After catching up to March 28 by merging upstream master into my fork, LT(5,KC_J) and similar keycodes stopped working on my FC660C (they still worked on my Planck EZ; I have a DZ60 I didn't test). Rather than momentarily activating the layer, I get stuck in that layer until I unplug and reconnect the keyboard.

While exploring this bug, I added an emergency escape DF(1) from every layer back to my base layer. These keys failed to work at all; I remained stuck in the 'momentary' layer.

Reverting the 27 upstream commits fixed the bug.

My keyboard is [fc660c syzygies](https://github.com/Syzygies/qmk_firmware/tree/syzygies/keyboards/fc660c/keymaps/syzygies)

My fork's git history should be public, making precise which commits I've fallen behind on.

## System Information

 - Keyboard:  FC660C with a Hasu controller
   - Revision (if applicable):  
 - Operating system: MacOS 10.14.6
 - AVR GCC version: avr-gcc (GCC) 8.3.0
 - ARM GCC version: 
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
 - QMK Firmware version:
0.8.79 (However I reverted before filing this bug report; this version works)
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [x] Other: Alfred 4.0.9 [1144]

## Additional Context

On request I can do a binary search to determine which of the 27 commits is responsible, and I can see if the DZ60 is affected. Usually I find my less obvious bugs by waiting a few hours and seeing if I slap my forehead. If someone's forehead slap doesn't resolve the bug, but suggests a commit to look at first, that would save me time.

I am a Kinesis Advantage survivor, so of course the first thing I considered was dropped USB events, but that wouldn't explain the escape DF(1) not working, or reverting the 27 upstream commits fixing everything. This is not sporadic behavior, it is 100% reproducible.

Erovia, 2020-03-29:
Hi @Syzygies, as I can see currently you are using `KC_NO` (XXXXXXX) on the temporary layers.
Could you do a test with `KC_TRNS` (_______) for me?

tzarc, 2020-03-29:
Struck the same thing on a test board locally -- KC_NO on target layer causes issue, KC_TRNS on target layer makes it work fine.

fauxpark, 2020-03-29:
When you use keycodes that activate a layer such as `MO()` or `LT()`, the keycode at the same spot on the target layer must either be `KC_TRNS` or the exact same keycode, otherwise you can lock yourself into that layer. This has always been the case, so perhaps there was actually a regression that has been fixed.

Syzygies, 2020-03-29:
I will make all tests suggested so far, including binary search to locate the commit responsible for this.

As for "the keycode at the same spot on the target layer must either be KC_TRNS or the exact same keycode" I did notice this in the doc, and I noticed that it had never applied to me before. Is there an explanation anywhere? Anti-bounce? 16 bits? If it was "fixed" by accident, can we fix it for real? If I write my own loop, will this still apply to me?

It's dumb luck / serendipity that my ANSI symbol layer needs 28 symbols, and there are 30 core keys in common for every keyboard I can imagine. I was using the two holes for redundant Backspace and Delete keys, but I can instead put the two holes under the tap/hold keys (in my case FJ) that access this layer.

I really don't want to punch a hole in my number layer. If this is inevitable, then the workaround is to separate each of my layers by hand. The right hand tap/hold key accesses the layer for left hand use, and vice versa. I would feel really dumb doing this, if it were easy for me to instead eliminate this restriction.

tzarc, 2020-03-29:
For reference, I bisected it, `5117dff6a26aec4eca04fb9787b4f428884739bc` is the first bad commit.

Syzygies, 2020-03-30:
So far I've used Git like a rock climber uses protection, except I don't fall so often. I just spent half an hour trying to locate this commit on GitHub, like a fly bouncing against a window on a hot summer day. Can't search by commit hash. Even dates are "ago" while my terminal gives precise time. I wanted to post a link to the commit in question, but that failed me as a novice. I have it in my terminal:

> commit 5117dff6a26aec4eca04fb9787b4f428884739bc
> Author: Drashna Jaelre <drashna@live.com>
> Date:   Sun Mar 22 06:29:05 2020 -0700
> 
>     Add Post Processing to process_record (#4892)
> 
>     * Improve process_record system
> 
>     Code based on @colinta's
> 
>     * Rename and better handle functions
> 
>     * Fix incorrect function call to process_record_user
> 
>     * Add documentation for post_process_record
> 
>     * Add both get_event_keycode and get_record_keycode functions
> 
>     And add some comments about these functions
> 
>     * Update code format
> 
>     * Cleanup merge artifacts

My suspicion from viewing this in my terminal is that the breaking changes are in post_process_record_user(), but I'm not sure, and I don't understand the code that well.

Syzygies, 2020-03-30:
> KC_NO on target layer causes issue, KC_TRNS on target layer makes it work fine.

I can also confirm this is my issue. Creating a pair of layers for each conceptual layer is a workaround. I'll try to figure out the exact code that is causing this, within the `5117dff6a26aec4eca04fb9787b4f428884739bc` commit.

fauxpark, 2020-03-30:
To elaborate on why this is not supposed to work, we have to consider how keycodes are resolved.

When a key is pressed, QMK searches through all active layers from highest to lowest. When, say, a `MO(1)` keycode is pressed from layer 0, layer 1 is not yet active, and so the keycode at that spot is ignored - the keypress is resolved as the `MO(1)`, which activates layer 1.

On keyup, the same thing happens, but this time since the layer is active, the layer scan hits that keycode. If it's a `KC_TRNS`, the scan continues downwards looking for the next non-transparent key, which should usually be the `MO(1)` on layer 0. If the keycode is anything else (or `MO(1)` as it will have the exact same numerical value), then the layer will never be deactivated because the keyup event that was resolved was for a different keycode.

Syzygies, 2020-03-30:
> To elaborate on why this is not supposed to work

Thank you, that helps. I've redone my layout after thinking this through. Soon I'll automate the duplication of layers, but for now I can get by while keeping up with QMK's upstream master. My [fc660c syzygies](https://github.com/Syzygies/qmk_firmware/tree/syzygies/keyboards/fc660c/keymaps/syzygies) keymap shows these changes.

It's important to me that my keyboards offer a base layer that is as safe as possible. Say, a student wants to borrow my office keyboard (not a current concern), and I don't want them accidentally rearranging the orbits of the planets. And I don't want to give up any keys to a dedicated TG(1) or similar, at least on a standard 60% or smaller keyboard: I migrated to QMK in the first place after frustration with an Anne Pro 2 for exactly this reason. Their firmware bungles tap/hold on their required internal function key.

So I was worried that because of this behavior, I'd be forced to give up a landing strip on my safety layer, to support switching back to it from advanced layers. Fortunately, this works: Hide a backdoor to one's function layer under the backspace key using LT(2,KC_BSPC), and put TG(1) somewhere on that function layer. This passes testing; I can go back and forth successfully while avoiding this issue.

So is this still a bug report, or a user whose luck ran out? That's a philosophical question, and I recognize that it's not my call.

I'd still like to understand the code for myself; if this is easy enough to fix that it temporarily got fixed by accident, my vote would be to adopt the fix. The behavior leads to less efficient layouts. I have a [GergoPlex: Heavy](https://www.gboards.ca/product/gergoplex-heavy) on order, but he has some _really_ small boards too.

Ideally, doesn't one want to cache the identity of a keypress so the keyup is interpreted the same way? Instead rescanning active layers _almost_ works, except for this issue. Does anyone actually view this as a feature? No one seems to have reported the bug while this behavior had accidentally changed. I do understand that older keyboards have very limited memories, and further complicated the code base is hard on everybody.

This behavior reminds me of my favorite pet peeve with all modern GUI operating systems: If the system takes a while to respond to a mouse event, it responds using the now-current context of the mouse event, rather than the context that the user saw while creating the event. I doubt that any operating system will get this right in my lifetime.

In any case, it appears that I can ultimately duck all of this by providing my own `process_record_user()`, which I plan to do in any case. Overloading this many alpha keys can get dicey, no matter how I set the standard tweaks for this.

mu-b, 2020-04-01:
Out of interest, how does this relate to the use of OSL? I have a LT(_LTRAN, KC_SPC) with OSL(_L2) set for key X on layer _LTRAN, hitting X again should invoke a macro on _L2, previously it would return to the original layer, this is now changed. I'm assuming that OSL semantics have now changed in line with the above in that I cannot use the key X on _L2 with now has to be KC_TRNS but instead some other key.

newtonapple, 2020-04-05:
I ran into the same issue.  Here is as simple use case I created: https://github.com/newtonapple/qmk_firmware/blob/newtonapple/layerbug/layouts/community/ortho_4x12/newton/keymap.c.  

I personally have few usecases where I put different `KC_*` codes under the `LT` keys.  For example, I have 2 different `LT`s that are pointing to the same layer.  I can alternate between the two `LT` keys to gain access to the keys hidden underneath them (see the code above).  Forcing `KC_TRNS` to be the only key code behind LTs seems a bit restrictive.

fauxpark, 2020-04-05:
> Forcing `KC_TRNS` to be the only key code behind LTs seems a bit restrictive.

Necessarily so. Think about it: if you hold an `LT()`, activating the target layer, what key is now in the "pressed" state? It *must* have the same keycode, otherwise, when you release it, the event that is fired will not deactivate the layer.
The "workaround" for this (really, the correct way to do it) is to create two near-identical layers, one with the transparent key over one `LT()`, and one with the transparent key over the other `LT()`.

Syzygies, 2020-04-05:
That was exactly my fix; I use a Ruby script to generate actual layers from a tab separated values file, cloning near-identical layers as needed. And I'm just about out of layers.

The firmware sees which physical key (xy-matrix) is pressed and released. The firmware could cache each physical xy-matrix location separately, saving the meaning of each key when it goes down, and using that meaning when the key comes back up, ignoring all layer changes that have happened since. It doesn't do this; the firmware instead rescans using the now-current context. That is an artifact inherited from FMK, not a law of physics.

And somehow this all got fixed by accident, for a while! I still don't understand the relevant code.

grassbl8d, 2020-04-05:
I also encountered this just now. The LT is permanently staying on that layer anymore. I thought I was doing something wrong as this has always been my configuration. My keyboard is kbdfans dz60 rgb ansiv2. 

Thanks.

drashna, 2020-04-05:
Oh, I think I see what the issue is ....

drashna, 2020-04-05:
Okay, narrowed down the issue.   An unintended, and odd ... one. 

Specifically, the cause is that this is getting called: 
https://github.com/qmk/qmk_firmware/blob/6de77141a4f7d45b19a148674beba1beb6711472/quantum/quantum.c#L165

It's updating the layer cache, it would appear, and causing the issue.  

Fortunately, this does appear to be a simple fix, at least. (tested already).

drashna, 2020-04-05:
For anyone running into this issue, could you test the above PR and verify that it fixes the issue? 

If you need help testing it, let me know.

Syzygies, 2020-04-05:
I have to step out, I'll finish testing in an hour, but for anyone else, here is how I managed to pull in this commit:

```
git remote add drashna https://github.com/drashna/qmk_firmware.git
git fetch drashna
git cherry-pick 175fb39c478785529c1136ff1f2b03ef48fca483
```

newtonapple, 2020-04-05:
@drashna I've tested this on my [test case ](https://github.com/newtonapple/qmk_firmware/blob/newtonapple/layerbug_fix/layouts/community/ortho_4x12/newton/keymap.c) as well as the Planck rev6, XD75 & Candy Bar.  The patch fixed the problem! 👍

Syzygies, 2020-04-06:
Thanks! We must be almost there. This commit also fixes the reported problem, but it introduced a new bug for me:

`TG(3)` works from any layer except `3` itself. `TG(4)` works from any layer except `4` itself.

(This worked before. My script that creates near-identical layers with transparent keys would accidentally work around this bug too. That was amusing/confusing!)

Here is a cached copy of my layout: [fc660c](https://github.com/Syzygies/log_folders/tree/master/2020-04-05)

I added TG(3) and TG(4) to layers 5 and 6 for test purposes, once I decided this must be the issue.

drashna, 2020-04-06:
> I added `TG(3)` and `TG(4)` to layers `5` and `6` for test purposes, once I decided this must be the issue.

That won't work.  Those toggle layers 3 and 4 on/off.  However, it reads "highest"/top down, so it will read 5 and 6 before 3 or 4.  So ... it may break that.  But ... I'll test TG in just a sec. 

@newtonapple and anyone else, if you could approve the PR in question (go to "files" in the PR and you can approve it there).  You won't get a green check (it will be grey), but it helps us know that it's tested and fixes the issue.

Syzygies, 2020-04-06:
> That won't work. Those toggle layers 3 and 4 on/off. However, it reads "highest"/top down, so it will read 5 and 6 before 3 or 4. So ... it may break that. But ... I'll test TG in just a sec.

I edited my previous comment in light of further testing.

To be clear, `TG(3)`  doesn't work when only layers 0 and 1 are active, and I'm using tap/hold F or J to temporarily access layer 3 in order to reach the `TG(3)`  on layer 3. The higher layers are disabled. `TG(3)`  _does_ work once one has layer 3 enabled by other means (such as the `TG(3)` key on layer 5, or by this bug before your fix).

drashna, 2020-04-06:
Could you post a link to your keymap, as well?

drashna, 2020-04-06:
I've made a couple more changes, and hopefully, that's still fixed (for those that applied my branch/PR)

Syzygies, 2020-04-07:
> Did this happen before the code change in question? (commit ID [5117dff](https://github.com/qmk/qmk_firmware/commit/5117dff6a26aec4eca04fb9787b4f428884739bc) )

As far as I can tell, this has never worked. I tested back to November 2019 (commit ID [e62ab7e25](https://github.com/qmk/qmk_firmware/commit/e62ab7e259d4f33c639a9433264c1cbfb6381cb5)). I couldn't go back much further without my builds breaking for what I assume are unrelated reasons. My setup may not be designed to go back years, for reasons I don't understand.

Should I file a separate bug report, or wait to see what you conclude?

The workaround is clear: Don't depend on `TG(3)` working on layer 3.

However, if the F and J keys stand in my mind as meaning the FJ layer, then using the F or J keys as part of locking that layer is easy to remember, and makes sense as language design. So it would be highly desirable to make this work. I don't see a workaround using the F or J keys.

Thanks!

EricGebhart, 2020-04-10:
I have the exact same problems, I will test tomorrow. I have add LT() on almost every key of my home row for many years. Luckily this is a new kb. Obviously trans is not the best solution.  I also have some osl() keys on the new xd75, those are on all layers via Trans, they do not work properly either.

foosinn, 2020-04-12:
Hey,

i just rebased the two commits of the open PR to my WIP split atreus.

Awesome work! I did not notice any issues. Thanks!

PR #8693 (2020-04-12): Fix bug with layer caching in get_event_keycode

## Description

Basically, the `get_event_keycode` handler was updating the layer cache when it shouldn't have been.  This was causing Layer Tap to get stuck, since it was no longer valid.

This adds an additional check to ensure if the layer cache should be updated or not. And updates all of the functions that are calling it (to be "false", so they don't update the layer cache, and instead read the layer cache.

## Types of Changes
- [x] Core
- [x] Bugfix

## Issues Fixed or Closed by This PR

* Fixes #8600

## Checklist
- [x] My code follows t [... cut]
closed 2020-04-12 by Erovia