qmk/qmk_firmware #8601: [Bug] Windows Setup Install fails with missing Paths

b14de, 2020-03-29:
## Describe the Bug
Following the setup instructions on https://docs.qmk.fm/#/newbs_getting_started?id=_1-download-software
The command `qmk setup` fails with:
```
INFO Found qmk_firmware at C:/Users/07837/qmk_firmware.
INFO QMK Doctor is checking your environment.
INFO Detected Windows.
ERROR Can't find arm-none-eabi-gcc in your path.
ERROR Can't find avr-gcc in your path.
ERROR Can't find avrdude in your path.
<class 'FileNotFoundError'>
ERROR [WinError 2] The system cannot find the file specified
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 592, in __call__
    return self.__call__()
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 597, in __call__
    return self._entrypoint(self)
  File "C:/Users/07837/qmk_firmware/lib/python/qmk/cli/doctor.py", line 226, in doctor
    bin_ok = check_binaries()
  File "C:/Users/07837/qmk_firmware/lib/python/qmk/cli/doctor.py", line 70, in check_binaries
    if not is_executable(binary):
  File "C:/Users/07837/qmk_firmware/lib/python/qmk/cli/doctor.py", line 156, in is_executable
    check = subprocess.run([command, '--version'], stdout=subprocess.PIPE, stderr=subprocess.PIPE, timeout=5, universal_newlines=True)
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
ERROR Your build environment is not setup completely.
*** Would you like to run util/qmk_install? [y/n] y
<class 'OSError'>
ERROR [WinError 193] %1 is not a valid Win32 application
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 592, in __call__
    return self.__call__()
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 597, in __call__
    return self._entrypoint(self)
  File "C:/msys64/mingw64/lib/python3.8/site-packages/qmk_cli/subcommands/setup.py", line 45, in setup
    process = subprocess.run(['util/qmk_install.sh'])
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 193] %1 is not a valid Win32 application

```
This error has been replicated on a second windows machine. Following manual installs of AVR GCC. 
ARM GCC and avrdude, setup continues to fail with error above 

avr-gcc --version and arm-none-eabi-gcc --version yield results as below.

No other errors occurred during the setup process described in the documentation.

## System Information

 - Keyboard:  
   - Revision (if applicable):  MDX-21292
 - Operating system: Windows 10 Pro
 - AVR GCC version: avr-gcc (GCC) 9.2.0

 - ARM GCC version: 9.2.1 20191025

 - QMK Firmware version: 0.8.80

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2020-03-29:
This has just been fixed in https://github.com/qmk/qmk_firmware/pull/8577. Please run `python3 -m pip install -U qmk` to ensure the CLI package is up to date, then run `git pull` from your `qmk_firmware` clone, and try again.

b14de, 2020-03-29:
Thanks @fauxpark, that worked as described with your suggested commands, much appreciated

closed 2020-03-29 by b14de