qmk/qmk_firmware #8610: [Bug] Matrix scan issues on Planck Rev 6 and Rev 6.1 on versions 0.7.6 and up

vomindoraan, 2020-03-30:
## Describe the Bug

Pressing certain keys on Planck Rev 6 and Rev 6.1 results in multiple presses being registered, as if the board had ghosting issues. This is not a hardware/electrical issue since it works fine on older versions of QMK.

For example, pressing Enter on the default keymap outputs `tg'`. It seems to be a matrix scanning issue.

[image]
Image courtesy of my friend who owns the Plancks.

After some investigation and testing on one of his boards, I've determined (by bisection) that the broken behavior was brought about by #6140. This PR removed the custom `matrix.c` files for Planck Rev 3 and Rev 6 since the DIP switch functionality contained therein was made a core feature. However, this apparently also messed with the scan/debounce timing (or something of the sort), which resulted in broken matrix scanning.

## System Information

 - Keyboard: Planck
   - Revision: Rev 6, Rev 6.1
 - Operating system: Windows
 - ARM GCC version: 6.3.1 20170620
 - QMK Firmware version: 0.7.6 and up

## Additional Context

This issue was also experienced by the OP of [this thread](https://www.reddit.com/r/olkb/comments/earda2/problem_flashing_new_planck_with_qmk_toolbox/) on the OLKB subreddit. Setting `DIP_SWITCH_ENABLE = no` as @drashna suggested in the thread didn't fix the problem.

Other affected boards may include Planck Rev 3 and Preonic Rev 3.

alex-ong, 2020-04-23:
(I don't have a planck)

Old debounce code; debounc_timer is part of scanning.
```
scan:
    scan cols:
        scan col
             compare col to previous scan / set debounce timer.
             store as columns in buffer

    check debounce timer:
        propagate to matrix, converting from columns to rows.
```

new:  Debounce timer only updated at end of scan.
```
scan cols:
    scan col, store as rows in buffer, compare col to previous scan (set bool if changed)
debounce(pass in bool):
   if diff, set timer.
   if timer expired, propagate to matrix
```

While they are not exactly the same, to my eye they should be equivalent.  The scanning code also looks equivalent-ish to qmk-cores default implementation. Only difference i can see is maybe how long it waits between pin-sets, working code uses 20us, new code (matrix.c) uses matrix_io_delay() which defaults to 30us. A longer delay shouldn't result in any weird noise chatter.

You could restore the custom_matrix and see if that works first; That way you have latest qmk_core but using old custom_matrix, hence excluding changes in qmk_core as part of the problem.

DonnieWest, 2020-06-11:
Just chiming in here that I am also experiencing this problem. Fresh clone of the qmk project resulted in this issue. If there's anything I can do to troubleshoot just let me know

DonnieWest, 2020-06-11:
I can confirm that switching back to the old matrix file for Planck Rev 6 resolves the issue for me

timshearer, 2020-06-22:
Think I've also had this issue on my Preonic Rev 3.  When pressing a particular key ("U" in my case) I would sometimes get no output and sometimes get multiple keypresses.  I switched back to the custom matrix and that seems to fix it (or at least it has so far).

speedRS, 2021-06-05:
@timshearer Having similar problems myself. Old post I know but any guidance on how to switch back to the old matrix?

timshearer, 2021-06-06:
@speedRS I managed to fix my issue but unfortunately I'm not 100% sure how! There is a possibility that it was hardware related though as I think it may have resolved after I reseated a switch. So perhaps an intermittent connection.

Sebas93cay, 2021-09-08:
Could anyone tell me how to switch back to the old matrix file for Planck Rev 6? 
Please and thank you very much

samjonester, 2021-09-13:
I found this thread after chatting in discord. My new planck/rev6 from Drop was registering phantom keypresses and behaving erratically. [Old firmware](https://github.com/noroadsleft/qmk_bits) worked flawlessly with the board. I was directed here and to [PR 6140](https://github.com/qmk/qmk_firmware/pull/6140) which removed a custom matrix file for the planck/rev6.

With the approach below, I was able to `qmk compile -kb planck/rev6 -km default && qmk flash  -kb planck/rev6 -km default`. I've also been able to compile and flash with minimal customizations to the default keymap with these steps.

1. From [PR 6140](https://github.com/qmk/qmk_firmware/pull/6140), I added the file [`keyboards/planck/rev6/matrix.c`](keyboards/planck/rev6/matrix.c) back into my qmk project.
2. I removed an include on line 8 for `backlight.h` which no longer exists.
3. I reversed the [diff](https://github.com/qmk/qmk_firmware/pull/6140/files#diff-a8484e03e69b21eeaf255d2fe897239480e0022c38bcfef74b71d9210c584d8b) to reference and enable the custom `matrix.c` file.
4. I ran `qmk compile -kb planck/rev6 -km default && qmk flash  -kb planck/rev6 -km default`
5. I tested with https://www.keyboardtester.com/
5. I ran `qmk new-keymap  -kb planck/rev6 -km <GH_USER>` and made a few customizations.
6. I ran `qmk compile -kb planck/rev6 -km <GH_USER> && qmk flash  -kb planck/rev6 -km <GH_USER>`
7. I tested with https://www.keyboardtester.com/
8. I rejoiced that `FT_T(KC_ENT)` gave me a right shift key AND an enter key without moving any keycaps!

This isn't a permanent solution. The custom matrix file is out-of-date and radically different from others in the project. However, it enabled me to successfully compile and flash my planck with [qmk v0.14.14](https://github.com/qmk/qmk_firmware/releases/tag/0.14.14). I think this suggests that the default firmware created with the default matrix file is incompatible with the latest hardware from Drop.

jackhumbert, 2021-09-19:
Thanks for all the insight here, folks. @samjonester, thanks for outlining your solution as well - @drashna has opened #14488 that adds an additional revision for the Planck & Preonic with these changes. [I've also uploaded the binaries here for quickly testing this solution](https://github.com/qmk/qmk_firmware/pull/14488#issuecomment-922390452). I don't have a whole lot of insight into why this has become a bigger issue right now, but it may have something to do with lower tolerances somewhere along the supply chain/manufacturing?

PR #14488 (2021-09-19): [Keyboard] Use old custom matrix for Drop Planck+Preonic
## Description

There have been numerous reports of the Drop OLKB boards having issues which are fixed by reverting to the old custom matrix.  Since it seems to be limited to Drop issues boards, added a new revision just for them.

Did not use custom matrix, as wanted to change as little as possible. 

## Types of Changes

- [x] Bugfix
- [x] Enhancement/optimization
- [x] Keyboard (addition or update)

## Issues Fixed or Closed by This PR

* Reported issues on discord, fixes #8610 

## Checklist

- [x] My code follows the code style of t [... cut]
closed 2021-09-19 by drashna