qmk/qmk_firmware #8611: USB2USB(Hasu) - Using keycode to enter bootloader

duoshock, 2020-03-30:
Is there a way to enter bootloader without using the physical switch on the converter?
I've tried reset keycode + bootmagic (space+b) that's didn't work.

Is it possible at all?

vomindoraan, 2020-03-30:
Both Bootmagic and Command are disabled by default on this converter. Try adding `COMMAND_ENABLE = true` to your `rules.mk` and resetting using LShift+LShift+B.

duoshock, 2020-03-30:
> Both Bootmagic and Command are disabled by default on this converter. Try adding `COMMAND_ENABLE = true` to your `rules.mk` and resetting using LShift+LShift+B.

Thanks for reply. I just tried it, with 'COMMAND_ENABLE = true' and 'BOOTMAGIC_ENABLE = yes', it reset the keyboard, but it won't stay in the bootloader for flashing,

Is there a way I can trigger the commend with simple keycode in layer? i.e. Fn2+Esc = enter bootloader.

duoshock, 2020-03-30:
I've also tried 'RESET' keycode, it's doing the same as above. :(

vomindoraan, 2020-03-30:
My bad, it should be `COMMAND_ENABLE = yes`, not `COMMAND_ENABLE = true`. But if the board stays in bootloader mode after being reset, it may be a quirk of the converter after all. Some ARM boards behave the same way and don't jump into the bootloader properly unless the physical reset switch is used.

duoshock, 2020-03-30:
That's a shame. I was thinking of making a custom usb2usb for Norbaforce. that will be very inconvenient to unscrew the backplate to flash.

Thanks for your help :)

vomindoraan, 2020-03-30:
Could you open QMK Toolbox and tell me what happens and what shows up in the console when you press LShift+RShift+B to reset the board while it's plugged in?

One final thing you could try is disabling Command, setting `BOOTMAGIC_ENABLE = lite` and adding the following to your `config.h`:
```c
// Location of the Esc key in the matrix (row 2, col 9)
#define BOOTMAGIC_LITE_ROW    2
#define BOOTMAGIC_LITE_COLUMN 9
```
Then compile and flash the board. Once you've done this, unplug the board and hold Esc while plugging it back in. If all goes well, it should go into bootloader mode and stay there.

duoshock, 2020-03-30:
Not sure did I do it right, It didn't recognise the converter with the default setting until I added 'CONSOLE_ENABLE   = yes'.
Once it recognised, I pressed LShift+RShift+B, then it printed 'Jumping to bootloader...', and no further message after that.
Windows detected there's a usb disconnected and reconnected within about a second, then the keyboard work as normal.

See screen cap here
[image]

After changed it to 'BOOTMAGIC_ENABLE = lite' and added theose 2 lines of code to keymaps\default\config.h
it's still doing the same - disconnected and connected soon after, didn't stay in bootloader.

zvecr, 2020-03-30:
Project seems to default to the caterina bootloader, which might not be correct in this instance. Tried `BOOTLOADER = atmel-dfu` in rules.mk?

duoshock, 2020-03-30:
> Project seems to default to the caterina bootloader, which might not be correct in this instance. Tried `BOOTLOADER = atmel-dfu` in rules.mk?

That's it! It works!
LShift+RShift+B works as intended.
Bootmagic to bootloader (space+B) works as well!

Thank you so much guys!

zvecr, 2020-03-30:
Raised #8613

PR #8613 (2020-03-30): Fix hasu usb converter bootloader

## Description

Verified by duoshock on #8611.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #8611

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [x] I have read the [**CONTRIBUTING** document](https://docs.qmk.fm/#/contributing). [... cut]
closed 2020-03-30 by zvecr