qmk/qmk_firmware #8641: ARM Cortex M-7 : STM32F745xx series

thewolf0, 2020-04-01:
Good Morning / good Afternoon, I want to know if the ARM Cortex M-7 especially the STM32F745xx series is compatible ? I know that is ''overkill'' but I just a simple question. I did some research and I found this https://github.com/qmk/qmk_firmware/blob/master/quantum/mcu_selection.mk : the MCU Cortex M-7 version isn't use.

Sorry if I wasn’t very clear or my mistakes in English.

yiancar, 2020-04-01:
I am pretty sure chibios supports it, so it should be quite easy to get it to work with a simple keyboard. If you want to do more complex things you will need to customize the drivers to support the peripherals of that chip

thewolf0, 2020-04-01:
All right, thanks for your answer, I’ll see this way

tzarc, 2020-04-01:
Yes, it's supported:

https://github.com/qmk/ChibiOS/blob/ver19.1.3/os/hal/ports/STM32/STM32F7xx/stm32_registry.h#L537

tzarc, 2020-04-03:
Feel free to open up a new issue if you've got specific problems with F745 support.

closed 2020-04-03 by tzarc