qmk/qmk_firmware #8655: Arch udev rules for flashing Pro Micro with qmk cli

NickyTope, 2020-04-02:
Hey, I have just succesfully flashed my Pro Micro based keyboard after much trial and error and would like to suggest an addition to the following page: https://github.com/qmk/qmk_firmware/blob/master/docs/faq_build.md 

In addition to the udev rules listed, I needed one for flashing..

```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="2341", MODE:="0666"
```

I added this rule to 52-tmk-keyboard.rules but it probably warrants having it's own file..

I found the device id needed by resetting the pro micro and immediately running `lsusb`

after this was added I could then flash with qmk cli, hoping this helps someone else as for all my searching I could only see people using `sudo make ...` as a solution

Erovia, 2020-05-12:
Thanks for the suggestions, I was just working on it.
My suggestion would (and will) be to add this rule to **etc/udev/rules.d/55-caterina.rules:**
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0036", TAG+="uaccess", RUN{builtin}+="uaccess"
```
This will add an ACL to the device for your user instead of making it world-writable.

NickyTope, 2020-05-13:
Looks great, thanks!

Erovia, 2020-06-07:
The docs have been updated.

closed 2020-06-07 by Erovia