qmk/qmk_firmware #8679: [Feature Request] New layout for Space65 - ISO, split spacebar, split backspace

TomTomDam, 2020-04-04:
Hello good people, I would like to request that a new layout for GrayStudio's Space65 keyboard is added which supports ISO, split spacebar and split backspace. This would be a very universal layout, as it accommodates the entirety of options that you can use for ISO users for this keyboard.

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior
- [x] New layout

## Description

Supported layouts for the Space65 (R2 Cybervoyager layouts are still the same as R1 Voyager):
[image]

Specifically, the layout that I am looking to do:
[image]

Features of the new layout should include:
- ISO
- Split spacebar (2.25u, 1.25u, 2.75u)
- Split backspace (1u, 1u)

I see that there is a LAYOUT_65_iso_blocker. however, it does not include split spacebar support for this specific layout. The default layout does support the split spacebar and backspace, except it does not have the big ol' Enter key and the adjacent #~ key. Therefore, this new layout will build on these existing layouts.

I'm excited to create this new layout myself, but right now I'm trying to familiarise myself with QMK, so code reviews would be great. I also don't have a Space65 of my own to test the layout, but if anybody is able to, please feel free to let me know how it goes.

Before starting work, I have a couple of questions:

- With the default layout, am I just able to map it as shown in the picture below to enable an ISO enter, so that I don't have to do this work in the first place? Bear in mind I don't have a Space65 or any other available PCB to test this out.
[image]

- Would it be better to make a more standardised layout, rather than one that's configured to my taste, for QMK? For example because I don't use the right Alt or Ctrl, I would prefer to replace them with Ins and Del. But it would be more convenient for the majority of people to have the right Alt and Ctrl to begin with.

Please let me know if this is feasible or not. Thanks guys.

vomindoraan, 2020-04-05:
That is correct, you can use the default layout to create an ISO keymap since it has all of the possible key positions mapped out. The only downside is that it doesn't appear with an ISO enter in the configurator, but it will work regardless of that. Although you might want to move `#~` to the ANSI `\|` position, and put `'"` where `#~` currently is.

TomTomDam, 2020-04-05:
> That is correct, you can use the default layout to create an ISO keymap since it has all of the possible key positions mapped out. The only downside is that it doesn't display an ISO enter in the configurator, but it will work regardless of that. Although you might want to move `#~` to the ANSI `\|` position (top enter key in your pic), and put `'"` where `#~` currently is.

Wow awesome, thanks for the help! It's good that the default layout has such flexibility. I will make the changes that you mentioned. Will close the ticket now.

TomTomDam, 2020-04-05:
> That is correct, you can use the default layout to create an ISO keymap since it has all of the possible key positions mapped out. The only downside is that it doesn't display an ISO enter in the [...]

Wow awesome, thanks for the help! It's good that the default layout has such flexibility. I will make the changes that you mentioned. Will close the ticket now.

closed 2020-04-05 by TomTomDam